devDucks/astroarch #3: Problema di connessione

AstroAlex1985, 2022-01-22:
Ciao,

ho scaricato AstroArch e flashato l'SD.

Collego correttamente il tutto all'hotspot, ma il link http://10.42.0.1:8080/vnc.html non funziona. Non si connette ad AstroArch.

E' percaso cambiato il link? 

Grazie

MattBlack85, 2022-01-22:
@AstroAlex1985 quindi vedi l'hotspot, ti ci colleghi ma non carica nel browser?

AstroAlex1985, 2022-01-22:
Ciao :-) Esattamente

MattBlack85, 2022-01-22:
@AstroAlex1985 posso chiederti una cortesia? Potresti collegarti alla rete del pi con lo smartphone dopo aver scaricato l'app "fing" (l'icona è una specie si cerchio ocn edei punti, bianchi su sfondo blu) fare uno scan della rete e riportare qui il risultato?

AstroAlex1985, 2022-01-22:
Ho provato. Al cellulare non si connette e Fing non vede la rete

MattBlack85, 2022-01-22:
@AstroAlex1985 credo di non aver sistemato il dhcp a dovere o di essermi dimenticato di abilitarlo durante il boot allora. Puoi provare ad assegnarti un indirizzo IP statico quando ti connetti all'hotspot, 10.42.0.100 tipo e dovresti riuscire a procedere.

Tornato dalle vacanze riguardo la parte del dhcp

MattBlack85, 2022-02-07:
@AstroAlex1985 ho rilasciato una nuova versione oggi che sistema il problema, scrivimi ancora se non dovesse andare nemmeno stavolta

closed 2022-02-07 by MattBlack85

AstroAlex1985, 2022-02-08:
Grazie mille :-) scarico subito