devDucks/astroarch #24: i2cdetect is not available

wbirk, 2023-06-11:
hi!
i cannot access i2cdetect, and i am too stupid to install it on arch :)
yours wolfi

MattBlack85, 2023-06-11:
hey @wbirk have you already updated to 1.6 through the command `update-astroarch`?

That was added in that version but the image to flash is the 1.5, the update shoudk bring it in

wbirk, 2023-06-12:
hi! that is what i was hoping for, yet it is not available :) yours wolfi

MattBlack85, 2023-06-12:
hey @wbirk that's weird, could you please report the output of the following commands?

`cat ~/.astroarch.version`

`pacman -Q i2c-tools`

wbirk, 2023-06-13:
hi!
version is 1.6
i2c-tools was not found ...
i am sorry for my total incompetence with arch ...
yours wolfi

MattBlack85, 2023-06-13:
> hi! version is 1.6 i2c-tools was not found ... i am sorry for my total incompetence with arch ... yours wolfi

don't need to be sorry, it's likely my (bad) bash skills :joy_cat: 

You can install i2cdetect by running `sudo pacman -S i2c-tools`

This will install the necessary bits @wbirk

MattBlack85, 2023-06-14:
@wbirk I found the issue, the update fails partially because I try to pull `indiserver-ui` which is a new package and `pacman` does not know yet about it.

I fixed that in mainstream but the upgrade to 1.5 failed and has not been applied, I suggest running by hand from the terminal `bash .astroarch/scripts/1.6.sh` to try it again

MattBlack85, 2023-06-29:
@wbirk I just released a new image ready to start on 1.6 (the link is on the README) directly where everything (you trust me? :joy: ) should work and everything is installed, gonna close this issue as i2c-detect it's already installed on that but please feel free to re-open should you have any issue

closed 2023-06-29 by MattBlack85