devDucks/astroarch #31: Passwords don't work

vidrazor, 2023-08-31:
I just installed AstroArch on an 8 gig Pi4, and when I try to run the updates or change location and time, it won't recognize the passwords. I've tried astro, astronaut, astroarch, AstrArch, nothing works.

So, what' sup?

Thanks.

vidrazor, 2023-08-31:
...also, I connected to the computer through a local network using VNC, and it never asked for a password, it just logged right in.

MattBlack85, 2023-09-01:
@vidrazor for the passwords not working, try to reboot, I got a couple of reports on that and rebooting usually solves it. Fr the VNC, there is no password to connect to it, that's by design, I believe you were expecting the password like astroberry does, but there is none on AstroArch

closed 2023-09-09 by MattBlack85