devDucks/astroarch #32: Astrometry Files Folder Is Locked

vidrazor, 2023-09-01:
The folder to copy astrometry files for plate solving is locked off, you can't download files, you can't copy files over, and you can't sudo dolphin over. The only choice is to copy an astrometry files folder to the desktop. 

WTF?

MattBlack85, 2023-09-01:
@vidrazor there is no locked off folders on the file system., you are likely trying to view/modify a folder `astronaut` user doesn't have access to (see, folders that belong to `root`)

The astrometry indexes are stored under `/home/astronaut/.local/share/kstars/astrometry/data`

vidrazor, 2023-09-02:
Thanks for your replies. I'm used to putting the astrometry data in .local/kstars/astrometry/data folder in Astroberry, so I couldn't understand why it was locked on this system. I have the astrometry files archived so there's so many, so I just copied the files to the pictures folder and pointed Ekos to it.

I re-flashed AstropArch because I had a Nikon driver failure that seem to stem from installing my Nikon right after I installed my Olympus, and the drivers apparently got crossed somehow and I kept getting a fail Nikon driver message, and had no control of my Nikons. After re-flashing AstroArch  I reinstalled the cameras shutting down KStars and reopening it each time and now all the cameras are recognized and running.

MattBlack85, 2023-09-02:
👍 glad it worked. I will add a mention to the path in the guide so if anyone else will look for the folder, it will be pointed in the right direction!

closed 2023-09-02 by MattBlack85