devDucks/astroarch #33: Chrome doesn't work

vidrazor, 2023-09-02:
Chrome does not come up when clicked on. I tried downloading a Debian installer on another Linux box, but it's just asking me what app to open it with instead of just installing. 

How can I install a new version, or another browser? I have no web access on the AstroArch system.

Thanks.

MattBlack85, 2023-09-02:
hi @vidrazor and thanks for this report! I never used chromium other than the time when I built this image originally (1+ year ago) and I tried now discovering there is an incompatibility with libre2.
This must be fixed upstream (by archlinuxarmmantainers) so I'd suggest in the meanwhile to `sudo pacman -S firefox` and have a go with it, probably I will swap chromium and FF in 1.7.

Thanks again!

vidrazor, 2023-09-03:
Thanks, OK, that worked, but how do you go about installing apps? There's no app manager program to install apps, and installers like Debian-based ones don't work.

Thanks again.

MattBlack85, 2023-09-03:
hi @vidrazor I will try my best to answer your question but feel free to ask for more precise things if this doesn't help.
Sooooo

Arch has a app manager, it is pacman but it's not graphical, it's a cli app (terminal based), pacman has full access to all applications packaged for ArchLinux, to see what application are available you can use this page https://archlinuxarm.org/packages

Now, those are the **official** packages but not all packages available to Arch, there are thousands of more packages in the so called **AUR**, AUR packages (you can find available packages here https://aur.archlinux.org/) are not official in the sense that are not mantained by the Arch packagers but instead are mantained by normal people like me.
Those cannot be installed with pacman but they need an alternative packager that knows how ti bundle them, my distro has  `paru` which knows how to follow the instructions to build an AUR packages, paru is a wrapper around pacman so it works the same (e.g. to install an AUR package you run `paru -S PACKAGENAME`.
It must be noted that AUR packages, while more often than not are functional, sometimes do not build cause may be outdated or no longer mantained.

There are some graphical wrappers for pacman, the most known are:

- pamac (paru -S pamac libpamac-full)
- octopi (paru -S octopi)
- tkpacman (paru -S tkpacman)

I suggest to use pamac, you can launch it after the installation with `pamac-manager`, you could try it and report back if you'd like.

For the debian installer I am not sure what you mean, clicking on .deb file and start the installation? Remember that `.deb` packages won't work on archlinux but the packages you mentioned are availale through pacman/paru

Be aware tho, but this is my personal dragons ahead, I never trusted graphical package manager in my life!

MattBlack85, 2023-09-03:
if it would be a nice addition, I could compile pamac and make it available to installation through pacman as I do with Kstars/INDI, and make it already available in my distro starting from version 1.7

vidrazor, 2023-09-05:
Thanks for your replies. 

I am what you may consider an "end-user". My goal is astrophotography via KStars/Ekos.

However I am not Linux savvy, I use Linux because KStars/Ekos doesn't work on Windows, not without jumping a lot of unwanted hoops, and I'm not a Mac fan. Astroberry and the Raspberry Pi has brought me into the Linux realm.

The whole terminal command-line thing in Linux is real pain for me.  I'll look into pamac.

I did not realize this was an alternative Linux distro, I thought it was some form of KDE. The graphic shell is odd in that it shows items in Dolphin against black with black type. It makes looking for things difficult, as can (hopefully) be seen in the screen grab below. Is there an alternative skin for this shell?

I simply want to install some apps, and in Linux, apps and driver installs always appear to be rituals to obtain. There's some basic apps missing here too. For instance, I was looking for a calculator, and realized there isn't one.

However for the objectives here (astrophotography), Stellarium and RawTherapee are useful resources to have on the system. The calculator would be nice too. :-) 

Thanks again for your replies. I appreciate what you've conjured up here with AstroArch, as Astroberry seems lost inthe 32-bit universe, and I believe the Pi can benefit from a 64 bit OS.

[image]

MattBlack85, 2023-09-09:
fair enough @vidrazor , I will compile and release pamac early next week so a simple `sudo pacman -S pamac` will be enough to install it. This will make the experience smoother!

For the zebra stripe, unfortunately some of the plasma packages broke during the last plasma update and it seems the breeze theme is one of those, for the meanwhile I's suggest o fallback to the default white theme, you can do so in Settings -> Appereance I believe

MattBlack85, 2023-09-25:
also @vidrazor it's been reported after latest update chrome is working FYI

MattBlack85, 2023-10-13:
closed as fixed in upstream

closed 2023-10-13 by MattBlack85