devDucks/astroarch #34: Is it possible to connect to Astroarch on a R-Pi by wifi, without hotspot?

stevecoh1, 2023-09-30:
The setup I want is to have my Raspberry Pi making a WIFI (not wired) connection to MY network and being accessible to my laptop, without having to change the WiFi connection to use the hotspot in AstroArch on the Pi, as if the Pi had a wired connection to the network.

Somehow, don't ask me how it works, I have such a connection working when the Pi is running Astroberry. Then, I can type http://astroberry.local/desktop/ into my browser and it goes right to the Astroberry desktop. I don't have to use the hotspot at all.  As I said, I have no idea how this works because the [Astroberry docs](https://www.astroberry.io/docs/index.php?title=Astroberry_Server#First_Boot) specifically state that you DO have to point the computer at the Astroberry's hotspot on the Pi.

> After the first boot, search for 'astroberry' wireless hotspot and connect your PC to it using 'astroberry' for password. Point your browser to https://astroberry.local/ or https://10.42.0.1/ to access Astroberry Server. For the Internet access connect your Astroberry to your home wireless network. Right-click top bar Wi-Fi icon and edit Wireless connection by entering your home network SSID and password.

This indicates to me that I should not be able to connect to astroberry without going through its hotspot, AND YET I CAN! I thought maybe this would somehow be cached in the browser but I just opened Firefox which I hardly ever use and it connected just as chrome does! If not in the browser, just how the heck does my computer find this address?

So I would like some networking guru to explain what might be going on.  If I understood how it worked in Astroberry, maybe I could get the same setup going with Astroarch. As of now, without a wired connection, I can connect to Astroarch only by connecting the laptop to the Astroarch hotspot and putting [http://10.42.0.1:8080/vnc.html](http://10.42.0.1:8080/vnc.html) in the browser.

Actually, the astroberry instructions above don't really work either. If I connect to the Astroberry hotspot, log in, and try to change the Pi's Wireless connection to my network, the connection from the laptop to the pi is lost and I can't do anything with it. And it's exactly the same with Astroarch.

So I am one confused puppy and hope someone can untangle all this and explain it. Thanks for listening.

MattBlack85, 2023-09-30:
Hi @stevecoh1

if I understand you correctly you just want the Pi to be connected to your place WiFi so that you can reach it using the browser without connecting to it directly through the Pi hotspot.

This is possible, and it works this way (both astroberry and astroarch), when you flash the image you start by default with the hotspot, this hotspot as a very low priority on the list of the networks but since there is nothing else, it is used.
Once you create a new connection (when you type your wifi credentials) it gets added with higher priority, so whenever _that_ network is available, it will be choosen over the hotspot.

When the WiFi is not available and there is no wired connection, the hotspot should be available again.

On AstroArch there is a small problem with the KDE wallet, so you need only one additional step, you must select "All users may connect to this network" like in the picture
[image]

After that the Pi should be connected to your wifi and you should be able to reach it using http://astroarch.local:8080/vnc.html

Let me know if it works

stevecoh1, 2023-09-30:
Thanks, a lot, Matt.  Yes, you have grokked what it is I am trying to do.  
However, one small problem.
Once I have logged onto the Pi, using the hotspot, I use the network GUI thing to select my home network.  Of course, this immediately aborts the noVNC thing as my connection is through the hotspot, but astroArch is now trying to connect to my network.  But the first thing it wants to know is my password, but I can no longer interact with the GUI.  Is there some text file I can encode this in, then reboot so it comes up in my network?

MattBlack85, 2023-09-30:
@stevecoh1 no unfortunately, but you can definitely try to disable the KDE wallet (yes, it is on my TODO's 🥲), the pass it wants it's coming from it so disbaling the wallet won't prompt for any pass.

To disable it (you can wire the pi just for this) you need to edit `/home/astronaut/.config/kwalletrc` adding the following lines:
```
[Wallet]
Enabled=false
```

you need to reboot to apply the change.

Should you have any trouble let me know!

stevecoh1, 2023-09-30:
But won't my network want that password to let me in?

stevecoh1, 2023-09-30:
Never mind the last comment.  I see the password was for the wallet.  
OK, so I saved the disablement in kwalletrc.  However, it's still no good. While connected to the Pi via the hotspot I connect the Pi to my network.  Of course, this disables the previous connection.  Back on my computer, I reconnect it to my Network and then try the URL http://astroarch.local:8080/vnc.html.  It does not work.
Is there some sort of latency that prevents this from working?

stevecoh1, 2023-09-30:
With a hotspot connection up, I was able to run the program nm-connection-editor.  Here, I was able to supply the password for my network.  This saved, and was readable upon reconnection.  But it still doesn't allow me to connect through my network.

stevecoh1, 2023-09-30:
Hmmm.  I guess there is latency involved. Now I can access Astroarch from the Pi on my computer.  Ta-daaa!

You might want to document this in your README.

stevecoh1, 2023-10-01:
OK, Matt, I think this is solved.  Basically it's a problem of documentation, both with Astroberry and AstroArch. Why **would** anyone want to connect to the Hotspot on the Pi? I can't think of a reason, although there could be one that I am missing.  Assuming I am right about this, _then the only reason you need the hotspot is for **bootstrapping** the Pi onto your network_.  You connect to the hotspot the first time, and can run the system.  But you really want the Pi to live on your network.  That is where the Astroberry instructions go wrong.  They tell you to use the network GUI widget to connect to the network, but as soon as that connection is made, then you can't do anything with the existing connection, i.e. the noVNC web page, which is connected through the hotspot.  I don't think the AstroArch instructions say anything about this. Therefore you need some other non-GUI way to tell Astroberry/AstroArch about your network. This is where the `nm-connection-editor` program comes in.  This program is part of the ArchLinux distribution.  (It's also part of the Astroberry Pi OS, and they might have mentioned it.)
You run that, add your network name and password, and you never have to use your hotspot again. Just reboot the Pi and it works. (But, fair warning, there is possibly a latency period of a few minutes before it can be used.) 

This is the sort of issue that might drive many potential users away from systems like AstroArch and ought to be documented.

MattBlack85, 2023-11-03:
this should be solved after the kwallet has been disabled by default in version 1.7

lhanson, 2024-06-08:
> Why **would** anyone want to connect to the Hotspot on the Pi? I can't think of a reason, although there could be one that I am missing. 

@stevecoh1 I'm guessing you only ever use your telescope at home, in range of your wifi, but that's not everyone's use case. My backyard is Bortle 7 so any chance I get I pack up my rig and head to a darker site. Once there, I use my laptop to connect to the hotspot and get everything running. I can then close my laptop and let astroarch run the imaging session while I do other things.