devDucks/astroarch #40: Impossibile connettere SynScan V4

AstroVersilia, 2023-12-14:
Salve Mattia,

Sto sperimentando AstroArch e non ho trovato nessuna difficoltà nell'installazione e nell'aggiornamento.
Provando però con due sysncan, mentre la V3 si connette senza problemi con Kstars, la V4 (che ha anche la porta usb) non c'è modo. 
Da cosa potrebbe dipendere?

Grazie,
Paolo

MattBlack85, 2023-12-14:
ciao @AstroVersilia, per capire meglio:

1. la synscan v4 funziona su altre versioni di astroarch o su astroberry/stellarmate?
2. connetti il telecomando al PC e la montatura al telecomando?
3. che montatura hai nello specifico?

AstroVersilia, 2023-12-14:
Salve Mattia,

La montatura è una eq6, con astroberry la v4 funzionava solo se connessa
con adattatore seriale ed impostata su connessione diretta con PC.
La V3 invece era sufficiente accenderla e dargli luogo e orario senza
opzione di connessione al PC.
Stessa cosa con AstroArch, la V3 è perfetta mentre la V4 non si connette.

AstroArch è perfetto per la mia camera, la Player One Artemis, invisibile
ad astroberry mentre AstroArch si è interfacciato perfettamente a tutte le
periferiche.

Grazie mille per l'interessamento e a presto

Paolo

AstroVersilia, 2023-12-14:
nel frattempo sono riuscito a far funzionare il synscan V4. Ho dovuto cambiare la porta della Artemis (usb3) e metterla su una usb2. Allora viaggia senza problemi. 
Questo problema c'era anche con Astroberry, che impediva il funzionamento del gps dongle se per la camera usavo la porta usb3.

Da cosa potrebbe dipendere?

Grazie e a presto.
Paolo

MattBlack85, 2024-02-14:
@AstroVersilia hai provato a cmbiare il baud rate per caso? forse cambiando porta e usando un baud rate basso (credo 9600) non riescono a dialogare, puoi provare 115200

AstroVersilia, 2024-02-15:
Buongiorno Mattia,

si ho fatto anche questa prova ma il problema non si risolve. Mi sa che
passerò al cavo diretto eqmod bypassando la pulsantiera

A presto e grazie
Paolo

MattBlack85, 2024-02-15:
@AstroVersilia se ne hai la possibilità secondo me è la cosa migliore. Con la eq6 ho sempre usato eqmod e si è sempre comportato bene. Su Linux in modalità eqmod con attacco diretto il baud 9600 non va comunque, devi forzarlo a 115200.

Fammi sapere!

MattBlack85, 2024-02-27:
questa la chiudo per ora!

closed 2024-02-27 by MattBlack85