devDucks/astroarch #41: GPIO access in AstroArch and Raspberry Pi 5

astrojolo, 2023-12-19:
Hello,

Is there any special access I need to have for GPIO chip in Raspberry Pi 5? 
I have installed https://abyz.me.uk/lg/lgpio.html library that works fine for me on RPi 5 and Debian based distributions. But I cannot make it connect to GPIO under AstroArch.

Steps I made:

1. fresh SD card flash
2. update-astroarch done
3. installation unzip, make, python and python3
4. installation lgpio from sources

After that, I ran rgpdiod daemon (started ok) and tried to connect to the GPIO chip with 

rgs c 1 go <chipnr>

with chipnr set from 0 to 5, and it always return -78 which means LG_CANNOT_OPEN_CHIP.

Many thanks in advance for any hits!

Lucas

astrojolo, 2023-12-19:
Ok, the issue was a permissions to GPIO. 
I have added a _gpio_ group, added _astronaut_ user to that group and added _gpio_ subsystem to that group as well using udev and it works now.

```
cat /etc/udev/rules.d/99-gpio.rules
SUBSYSTEM=="gpio", KERNEL=="gpiochip*", GROUP:="gpio", MODE:="0660"
```

MattBlack85, 2023-12-19:
@astrojolo thanks 🔥 will leave this around for troubleshooting

PS: I packaged astrolink4pi for Arch, I don't have your hardware yet but could you tell me if installing it with pacman leads to a working setup for your board?

Pozdro

astrojolo, 2023-12-19:
I have added a section of additional tasks to make my hardware work with AstroArch:
[](https://github.com/astrojolo/astrolink4pi?tab=readme-ov-file#astroarch-only-specific-tasks)

It now works with revision 4 of hardware, however for revision 3 I need to use an SPI device, and at the moment, I cannot communicate with it even after creating a proper udev entry. I need to investigate it.

astrojolo, 2023-12-19:
Ok, all works fine. I needed to enable SPI in /boot/config.txt:
`dtparam=spi=on
`
All required steps are listed on [https://github.com/astrojolo/astrolink4pi?tab=readme-ov-file#astroarch-only-specific-tasks](https://github.com/astrojolo/astrolink4pi?tab=readme-ov-file#astroarch-only-specific-tasks)

Thanks and congratulations on this distribution!