devDucks/astroarch #42: Missing Toutptek drivers

teoteo, 2023-12-21:
Updating from AstroArch 1.7 to Kstars 3.6.8 and INDI 2.0.5 INDI 2.0.5 [indi_toupcam_ccd](https://www.indilib.org/ccds/toupcam-cameras.html) driver is missing and I get this error:

`[2023-12-14T17:18:53.444 CET WARN ][           org.kde.kstars.indi] - Driver "Touptek" failed to start. Retrying in 5 seconds...
`

[image]

stevecantlay, 2023-12-21:
same issue when trying to reinstall drivers and lib 

pacman -S indi-3rdparty-drivers
pacman -S indi-3rdparty-libs  

getting the following error on build:

==> Generating module dependencies
==> Creating gzip-compressed initcpio image: '/boot/initramfs-linux.img'
==> WARNING: errors were encountered during the build. The image may not be complete.
error: command failed to execute correctly

teoteo, 2023-12-22:
May be it is a related issue https://github.com/indilib/indi-3rdparty/issues/863?

stevecantlay, 2023-12-22:
Possibly, I built the drivers, and librarys from source this morning and all seem to be happy

MattBlack85, 2023-12-22:
seems my building machine had a small blip (and TBH I have nfi why), rebuilt the drivers and touptek is there. I am rebuilding a new version of indi-3rd-party-drivers. Should be released in 20 minutes

MattBlack85, 2023-12-22:
@teoteo @stevecantlay `indi-3rdparty-drivers` version `2.0.5-2` is available now, I see there `indi_toupcam_*` binaries so I think everythng will be ok. Will clsoe this issue as fixed but feel free to re-open it if there are still issues

closed 2023-12-22 by MattBlack85