devDucks/astroarch #43: Problema raffreddamento camera ToupTek

Lukadi80, 2023-12-30:
Salve Mattia, 
E complimenti per questo progetto che stai portando avanti!
Ho un problema con la mia camera ToupTek ATR3CMOS09000KPA con sensore imx 533 con ekos su astroarch. Il comando di raffreddamento non funziona; inserisco la temperatura desiderata ma non si avvia il raffreddamento anche se la ventola è in funzione. Ho provato la camera con NINA e funziona perfettamente. Come si può risolvere? Sarà un problema del driver?
Grazie

MattBlack85, 2023-12-30:
Chiediamo a Teo che usa le toup pure. @teoteo da te il raffreddamento funziona senza intoppi?

@Lukadi80 che versioni di indi hai al momento?

Lukadi80, 2023-12-30:
Ieri sera per la prima volta ho installato la disto e ho fatto subito l'aggiornamento come da guida! E poi ho avviato il tutto...all'inizio avevo dei problemi con le porte; montatura e focuser vedevano la stessa porta...ma poi ho risolto! Pertanto in riguardo la tua domanda, penso che sarà l'ultima  versione indi disponibile !

MattBlack85, 2023-12-30:
Ok allora è la più recente, vediamo cosa dice Matteo

teoteo, 2023-12-30:
> Chiediamo a Teo che usa le toup pure. @teoteo da te il raffreddamento funziona senza intoppi?

L’ho collegata ora e sì, il raffreddamento funziona. Qui il log
[log_11-43-54.txt](https://github.com/devDucks/astroarch/files/13797802/log_11-43-54.txt)

https://github.com/devDucks/astroarch/assets/789650/0223e13b-e4bb-415b-b96d-54d39be6bb33

Lukadi80, 2023-12-30:
Quindi la causa quale potrebbe essere considerato che sotto i programmi windows con ascom funziona?

teoteo, 2023-12-30:
Nella parte inferiore del pannello indi della camera, quando premi sul pulsante per attivare il raffreddamento compare qualcosa come log?

MattBlack85, 2023-12-30:
Inoltre ci puoi dire come hai cablato il tutto?

Lukadi80, 2023-12-30:
Prima di rispondere ho voluto fare diverse prove  e alla fine sembra funzionare ma a volte fa dei capricci...vi spiego! Appena ho accesso oggi per la prima funzionava! Però nel pannello Indi vedevo che c era una temperatura di default con valore 100...Appena l'ho modificata e portata a 0 gradi di botto si è arrestata! Poi uscendo da kstars e rientrando riporta sempre il valore 100 ma funziona!... poi modificando il valore sulla scheda di ekos portava la temperatura dive desideravo! Nelle diverse prove mi è successo che dopo un pò si è spenta da sola e spegnendo e accendendo dai tastini della scheda un paio di volte, si riaccendeva! Insomma, non è molto stabile...forse va rivisto il driver? Farò altre prove nei prossimi giorni e vediamo! @MattBlack85 ti segnalo anche degli arresti anomali della disto...a volte anche senza programmi aperti oppure nel mentre che sta aprendo Kstars! @teoteo considerato che usi la touptek se mi puoi consigliare i valori ottimali di gain e offset che utilizzi con ekos! Grazie del pronto supporto e complimenti ancora del lavoro che porti avanti con astroarch!

teoteo, 2023-12-30:
Il fatto dell’avere sempre impostati di default 100 gradi credo sia una cosa definita da Touptek nel driver. Fra l’altro anche salvando i propri parametri quello si riporta sempre a 100.
Su gain e offset siamo OT. Comunque di solito uso 100 di gain per UV/IR-Cut e 800 per L-Ultimate e 1310 di offset. Per calcolarlo ad hoc, [guarda qui](https://www.youtube.com/watch?v=mCYQoVSuEaI).

Lukadi80, 2024-01-17:
Buongiorno @MattBlack85! Ritorno a disturbarti con lo stesso problema! La camera ha continuato ad avere problemi con il raffreddamento; devo disconnettere e connettere più volte affinché si attivi...e a volte capita che si blocca! Ho esposto questo problema all'assistenza touptek e loro mi hanno chiesto se nella libreria indi di kstars sono caricati gli ultimi aggiornamenti dell'sdk toupcam recentemente messi a disposizione su git hub. Ti chiedo gentilmente se puoi verificare così mi confronterò con loro se c è un difetto di fabbrica oppure c era una mancanza di aggiornamento. Grazie!

Lukadi80, 2024-01-17:
Grazie Mattia...intanto comunicherò questa situazione a touptek... è anche possibile che anche con l'ultimo aggiornamento rilasciato da loro non si risolve nulla! Di nuovo grazie

Lukadi80, 2024-02-06:
> @Lukadi80 ho appena rilasciato le nuove versioni di indi e kstars se vuoifare una prova!
Ciao Mattia...grazie per l'info!! alla fine ho risolto era un problema di alimentatore...adesso va benissimo! ma stasera provvederò ad aggiornare il tutto! rinnovo sempre i miei complimenti perchè hai creato una disto che va benissimo e mi trovo meglio rispetto ad astroberry! un saluto...

MattBlack85, 2024-02-06:
allora qui chiudo!

closed 2024-02-06 by MattBlack85