devDucks/astroarch #44: unable to set time

BenoitBotton, 2024-01-24:
I am  trying my hand at using Kstart/Ekos to trigger my camera and was very happy to find your fantastic package

however, I am unable to set the time and date on my brand spanking new astroarch installation, no issue changing the timezone, but it has not updated the clock, the system thinks we are 6 days ago!

[image]

my pi4 B is connected to my wifi network, so I would have expected it to update time after three or for reboots by now.

if I untick 'set date and time automatically' update and apply, I get 'unable to set current time' after entering the password when requested:
[image]

any idea why that might be?

BenoitBotton, 2024-01-24:
following https://wiki.archlinux.org/title/System_time, I was able to set the time using `timedatectl set-time "yyyy-MM-dd hh:mm:ss"` however this is rather crude in terms of accuracy so decided to setup SNTP using systemd-syncd but so far the system clock is not synced

MattBlack85, 2024-01-26:
hi @BenoitBotton you mean that even if you are connected to the internet your time doesn't sync automatically? This is weird as it should sync immediately on boot whenever the wifi/eth link is active. Can you check what the NTP service is telling you kindly?

BenoitBotton, 2024-01-27:
Hi @MattBlack85 ,
Yes, when I connect and wait, the time does not sync. I booted yesterday and despite setting up sntp [other day, no sync happened. The Pi actually thought we were a day earlier.
I will spent some time on it later this weekend I'll be able to give you the info you asked then

BenoitBotton, 2024-01-27:
I am getting the following:
```
~ » timedatectl                                                                                
               Local time: Wed 2024-01-24 22:04:13 GMT
           Universal time: Wed 2024-01-24 22:04:13 UTC
                 RTC time: n/a
                Time zone: Europe/London (GMT, +0000)
System clock synchronized: no
              NTP service: active
          RTC in local TZ: no
```
 when it is 27/01/2024 11:46 (Bristol, UK)

```
timedatectl show-timesync --all                                                       130 ↵ astronaut@astroarch
LinkNTPServers=
SystemNTPServers=0.arch.pool.ntp.org 1.arch.pool.ntp.org 2.arch.pool.ntp.org 3.arch.pool.ntp.org
RuntimeNTPServers=
FallbackNTPServers=0.pool.ntp.org 1.pool.ntp.org 0.fr.pool.ntp.org
ServerName=
ServerAddress=
RootDistanceMaxUSec=5s
PollIntervalMinUSec=1d
PollIntervalMaxUSec=1month 1d 13h 30min
PollIntervalUSec=0
Frequency=0
```
I guess empty ServerName and SeverAddress are a hint!

BenoitBotton, 2024-01-27:
just to clarify, should I need to setup systemd-syncd? 
I am wondering whether to restart the whole thing from burning a new image as it is very early days for this system, there's only one Ekos config file to backup

BenoitBotton, 2024-01-27:
so I went back and reflashed my SSD.
on first boot did `sudo timedatectl set-timezone Europe/London`
then connected the Pi to my wifi and ....
```
timedatectl                                                                                                                                            astronaut@astroarch
               Local time: Sat 2024-01-27 14:06:35 GMT
           Universal time: Sat 2024-01-27 14:06:35 UTC
                 RTC time: n/a
                Time zone: Europe/London (GMT, +0000)
System clock synchronized: yes
              NTP service: active
          RTC in local TZ: no
```

I am guessing that my original attempt to change timezone using the GUI was the culprit. I will leave this open for now as I am still to 
- update the system
- set swap
- resize partition

I will confirm time is still syncing after the above and a couple of reboots

MattBlack85, 2024-01-27:
from what I see it seems that you disable the system sync (in your old report it shows no while in the new one it shows yes) somehow and I believe that was why the time wasn't syncing even with an active NTP.
That is my best guess at least

BenoitBotton, 2024-01-28:
yes, probably. I think my mistake was to try and set the timezone and time from the UI. so in the process I did uncheck the 'set time automatically box' which would have stopped the syncing. but then it did not restart when I ticked the box.

anyway, time is still syncing so I am closing this.

thank you very much again for your help and more importantly for your work making astroarch available to the community

closed 2024-01-28 by BenoitBotton

MattBlack85, 2024-01-28:
thanks to you too! Reports only make things better and serve the others for documentation!