devDucks/astroarch #48: Configuring GPS in Astroarch

sc74, 2024-02-19:
A little code idea to insert into the astroarch_build.sh file for GPS. The # Set GPS part can also be a file in the .astroarch/configs folder to be uploaded to /etc/default/gpsd

# Set GPS
rm /etc/default/gpsd
touch /etc/default/gpsd
cat > /etc/default/gpsd << 'EOL'
#Default settings for gpsd.
START_DAEMON="false"
GPSD_OPTIONS="-n"
DEVICES="/dev/ttyACM0"
USBAUTO="false"
EOL

# 2 lines to synchronize the system time with the GPS
echo 'refclock SHM 0 offset 0.5 delay 0.2 refid NMEA' >> /etc/chrony.conf
echo 'driftfile /var/lib/chrony/drift' >> /etc/chrony.conf

# Start GPS
systemctl enable gpsd
systemctl start gpsd

sc74, 2024-02-19:
To use xgps, you must install the python-cairo package

Thank you

Stephane

MattBlack85, 2024-04-10:
fixed in 87874fd1c55fedbf3f1c65f4a5fa9eb3bf76aff6

closed 2024-04-10 by MattBlack85