devDucks/astroarch #50: Unprotected VNC server

sc74, 2024-02-19:
The VNC is not password protected. 

To protect yourself when several networks are nearby?
Think this is useful?

THANKS.

Stéphane

A small line in astroarch_build.sh

# Password tigervnc
(echo astroarch; echo astroarch; echo y; echo astroarch; echo astroarch) | vncpasswd

And this in X0vncserver.service

ExecStart=x0vncserver -display:0 -rfbauth /root/.vnc/passwd

MattBlack85, 2024-02-26:
hey @sc74 

as AstroArch wants to be simple, providing a default password for vnc won't add an additional layer of security as it is freely available on the internet.
On the other side, documentation is crying loudly cause I haven't cared too much about it, for such a case it should be more than enough to add a section about this in case someone wants to harden the setup

sc74, 2024-02-26:
Actually documenting it is a good thing, it could help people who are in areas with high concentrations of wifi. 

I also thought about the fact that several rpis, PCs running AstroArch can be nearby (I use 3 during my astro outings). And that this could avoid confusion if the documentation also explained how to change the ssid of the wifi network.

MattBlack85, 2024-03-16:
@sc74 if a user gives `vncpasswd` command as astronaut, the pass will be stored in `/home/astronaut/.vnc/passwd` correct?

sc74, 2024-03-16:
This is correct for user.

I implemented this to add the root password in x0vncserver.service:

ExecStart=x0vncserver -display:0 -rfbauth /root/.vnc/passwd

And done, sudo vncpasswd to get the password in /root/.vnc/passwd

MattBlack85, 2024-04-12:
done in 8c1057db8b0e77cf84b2a83026c0c68ad01d237c

closed 2024-04-12 by MattBlack85