devDucks/astroarch #51: Useful packages for Arch user

sc74, 2024-02-19:
Some useful packages for users, maintenance or future versions of AstroArch.
System:

rsync wget parted dosfstools sudo accountsservice  cmake meson opencv ffms2 libgit2 vim ninja gtk3 ark irqbalance inxi lsb-release ncdu plymouth smbnetfs alacritty

Kde:

kdenetwork-filesharing dolphin-plugins qt5-imageformats

THANKS

Stéphane

MattBlack85, 2024-03-06:
Out of curiosity @sc74 Alacritty was once the official terminal in AA but I had to nuke from orbit, it was a CPU hog. How is behaving recently?

sc74, 2024-03-06:
I have to say that I don't remember why exactly I installed it. In any case, it has never posed a problem with the CPU or memory. You can without going through for the moment and install it if necessary.

sc74, 2024-03-07:
I found for I had alacritty. I tested wayland with sway without result.

MattBlack85, 2024-04-12:
most of those packages are for building other packages and won't be part of the base image (rsync btw is already available to be installed via pacman as I pre-compiled it), this leads to me to think if someone wants to build packages on AstroArch likely knows what's doing and how to install them :D

closed 2024-04-12 by MattBlack85

sc74, 2024-04-12:
You're absolutely right. This topic will probably be discussed later. In particular, during special astro user requests. Especially, if they are not interested in creating their packages and for whom it will be necessary to provide the simplest user experience possible ;)

This issue can be closed.