devDucks/astroarch #52: Astro software

sc74, 2024-02-19:
Good morning,

Do you plan to keep collimation-circles?

It is practical for roughing out the collimation of a sc tube.

A software that is missing and which seems important to me given its evolution is Siril.

Indeed, this software is not too heavy for a rpi.

It allows you to quickly open a fit image and work on it. It also allows you to do live stacking.

But the most essential thing for me is that it allows you to visualize the faults of the instrument (tilt, backfocus, etc.). It already has good tools for here and there, future version will bring others. It helps me a lot during my outdoor astro sessions.

Good clear sky.

Stéphane

sc74, 2024-02-20:
The Aur package Aladin could also be another interesting software that is little seen on astro distributions.

It allows you to visualize sky maps by superimposing them on astronomical data. Very practical for discovering all the objects present in our images.

teoteo, 2024-02-24:
> Do you plan to keep collimation-circles?

I think the answer lies in [this issue](https://github.com/sajmons/CollimationCircles/issues/19).

sc74, 2024-02-24:
Thank you teoteo

Let's wait for the new version

clear sky