devDucks/astroarch #53: FTP server?

sc74, 2024-02-21:
Maybe it would be useful to have a ftp server such as vsftpd on ArstroArch? What do you think ?

sc74, 2024-02-21:
If you decide to include FTP here is a suggested code once the package is installed.

sed -i 's/#write_enable=YES/write_enable=YES/g' /etc/vsftpd.conf
sed -i 's/#local_enable=YES/local_enable=YES/g' /etc/vsftpd.conf
sed -i 's/#ascii_upload_enable=YES/ascii_upload_enable=YES/g' /etc/vsftpd.conf
sed -i 's/#ascii_download_enable=YES/ascii_download_enable=YES/g' /etc/vsftpd.conf
sed -i 's/#chroot_local_user=YES/chroot_local_user=YES/g' /etc/vsftpd.conf
sed -i 's/#chroot_list_enable=YES/chroot_list_enable=YES/g' /etc/vsftpd.conf
sed -i 's/#chroot_list_file=/etc/vsftpd.chroot_list/chroot_list_file=/etc/vsftpd.chroot_list/g' /etc/vsftpd.conf
sed -i 's/#ls_recurse_enable=YES/ls_recurse_enable=YES/g' /etc/vsftpd.conf
cat > /etc/vsftpd.conf << 'EOL'
# if not specified, users' home directory equals FTP home directory
local_root=public_html
# turn off seccomp filter if cannot login normally
seccomp_sandbox=NO
EOL
touch /etc/vsftpd.chroot_list
cat > /etc/vsftpd.chroot_list << 'EOL'
# add users you allow to move over their home directory
astronaut
EOL

Good day.

Stephane

MattBlack85, 2024-02-26:
hi @sc74 

I think an FTP server may be of interest for some people, but definitely I think is more of an edge case.
For these kind of things, I think the best solution is to provide clear documentation on how to achieve it in case you need it.

AstroArch grew quite a lot in the number of packages that are present by default and extra care should be taken to avoid adding more (together with dependencies) that are rarely used.

I will mark these kind of issues with the `add documentation` label

sc74, 2024-02-26:
I understand. This is a suggestion, because I saw that Jasem had implemented it in Stellarmate and that he suggested that some people use it when transferring large files such as astrometry files. 

My proposal is still taken from the Arch wiki https://wiki.archlinux.org/title/Very_Secure_FTP_Daemon

MattBlack85, 2024-04-12:
this will be a zsh plugin

MattBlack85, 2024-06-11:
fixed in 1.9

closed 2024-06-11 by MattBlack85