devDucks/astroarch #54: Warning at kernel startup

sc74, 2024-02-22:
Hello,

Some small fixes for kernel startup.

1 - Modification of the /etc/fstab file to eliminate the "failed to start remount root and kernel file systems" message

PARTUUID=cafd14b5-02 / ext4   defaults    0 1

2 - Error message "at-fs not properly umounted". To be checked when creating the image. Solved after installing dosfstools package and doing fsck.vfat /dev/sda1

3 - avahi asks for the nss-mdns package to be installed. Install the package?

4 - A line is missing in the service file x0vncserver.service so that it starts after the network.

After=network-online.target

A great day.

Stephane.

MattBlack85, 2024-02-26:
will modify the fstab and have a look at what I have done with my vfat partition.

3 - nss-mdns is an optional dependency, it _shouldn't_ be needed for any normal mDNS case, we should be good without it
4 - I removed the line on purpose, people may take the Pi in the wild and there will be no internet connection, that way the VNC server will start after 90s I believe as it is the timeout for waiting, I don't think we should wait till the network has reached the online status before starting VNC, unless this may cause troubles where VNC cannot start because the network is not online

sc74, 2024-02-26:
Ok for 3

For point 4, I make this remark in order to have a clean start in the logs because indeed vnc does not like the fact that the network is mounted before it. This doesn't cause a problem, just vnc crashes and the Restart=on-failure line restarts the service.

Besides, this restart is what happened when X11 crashed because of 3D acceleration in 99-v3d.conf.

MattBlack85, 2024-03-06:
these have been fixed in 1.8

closed 2024-03-06 by MattBlack85