devDucks/astroarch #55: Unstable VNC server

sc74, 2024-02-24:
Hi,

The 99-v3d-conf file must be modified like this to disable 3d acceleration which is experimental in v4c. Because this creates outages in the vnc service.

“OutputClass” section
  Identify "vc4"
  MatchDriver "vc4"
  Driver “modesetting”
  "AccelMethod" option "none"
EndSection

A clear sky

Stephane

teoteo, 2024-02-26:
I tried it but I get an error with _"AccelMethod" option "none"_ but I get startup display error
With _[Option "Accel" "Off"](https://man.archlinux.org/man/extra/xorg-server/xorg.conf.d.5.en)_ works but how I can test if there are still vnc outages?
```
~ » cat /usr/share/X11/xorg.conf.d/99-v3d.conf
Section "OutputClass"
  Identifier "vc4"
  MatchDriver "vc4"
  Driver "modesetting"
  Option "PrimaryGPU" "true"
  #"AccelMethod" option "none"
  Option "Accel" "Off"
EndSection
```

sc74, 2024-02-26:
Oops, it's my fault
I reversed the code. It's not "AccelMethod" option "none" but :

option "AccelMethod" "none"

There are only two options for "AccelMethod", "glamor" or "none" https://man.archlinux.org/man/modesetting.4
The disconnection problems were clearly visible in Ksystemlog and systemctl status x0vncserver
What wasn't very clear at first was whether it came from vnc or X11

closed 2024-02-26 by MattBlack85