devDucks/astroarch #59: Raspberry PI 4, automatically start the cooling fan.

oldmar, 2024-02-28:
Raspberry PI 4
with Debian or Ubuntu distributions I typed from the console "sudo pip3 install --no-cache melopero-fan-hat" to automatically start the cooling fan.
Question: What command to type in archlinux (astroarc) konsole to automatically start the cooling fan?
Thank you

MattBlack85, 2024-03-03:
Hi @oldmar can you try `pip install --user --no-cache melopero-fan-hat `?

oldmar, 2024-03-03:
Can you be more precise and detailed in suggesting the unique command to insert?
Does using AstroArch involve knowing everything about this distribution?
Thank you

pip --version                                                                     1 ↵ astronaut@astroarch
pip 24.0 from /usr/lib/python3.11/site-packages/pip (python 3.11)

---------------------------------------------------------------------------------------------------

~ » pip install --user --no-cache melopero-fan-hat                                    2 ↵ astronaut@astroarch
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try 'pacman -S
    python-xyz', where xyz is the package you are trying to
    install.

    If you wish to install a non-Arch-packaged Python package,
    create a virtual environment using 'python -m venv path/to/venv'.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip.

    If you wish to install a non-Arch packaged Python application,
    it may be easiest to use 'pipx install xyz', which will manage a
    virtual environment for you. Make sure you have python-pipx
    installed via pacman.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
--------------------------------------------------------------------------------------------
pip install --astronaut --no-cache melopero-fan-hat                                   astronaut@astroarch

Usage:   
  pip install [options] <requirement specifier> [package-index-options] ...
  pip install [options] -r <requirements file> [package-index-options] ...
  pip install [options] [-e] <vcs project url> ...
  pip install [options] [-e] <local project path> ...
  pip install [options] <archive url/path> ...

no such option: --astronaut

MattBlack85, 2024-03-03:
hey @oldmar not sure why you think I gave you a chunked command, that was the raw full command.  --user, which has nothing to do with Arch, tells python to install packages under the local user directory, thought --user still works but it seems it doesn't. 

Make no mistakes, this is not because of ArchLinux, this is pip behavior, after PEP668 https://peps.python.org/pep-0668/ OSes are upgrading to satisfy the proposal. Arch has onboarded that a long ago (but others will follow ad well at some point), what it means is that you cannot install python packages globally (sudo) by mistake. 
To install those you have 4 ways:
- install from the official packages (sudo pacman -S python-XXX) 
- install from AUR (paru -S python-Xxx) 
- if the packages are not there ask the mantainer to make one (it is me on this case) so they can be installed with paru/pacman
- use the --break-system-packages pip flag (this may create some conflict issue with other packages tho)

MattBlack85, 2024-03-03:
to look for packages on AUR https://aur.archlinux.org/packages

to look for packages from official repos https://archlinuxarm.org/packages

MattBlack85, 2024-03-03:
> I understand, AstroArch program creates too many difficulties to manage the cooling fan installed in the Raspberry.
> Managing Archlinux system packages is out of my interest in astronomy, maybe I shouldn't have wasted time with this program and not waste your time replying.
> Thanks for the support

Sorry, but that's has 0 to do with what I said and 0 to do with Arch.

People have asked me to create packages not available to be installed since the beginning of this and it's never been an issue.

I will try to package the melopero stuff as it seems more people may want to use it in the future, the creator has a weird way to uninstall the package tho and that may require a lil more time but as every python package already available, it shouldn't be that big of an issue

MattBlack85, 2024-03-03:
Reference for myself => https://github.com/devDucks/astroarch-pkgs/issues/11

oldmar, 2024-03-03:
Mine was just a question, I'm not asking anything.
I wrote only to understand if with a simple command from the AstroArch terminal, the problem of starting the cooling fan at system boot could be solved, nothing more.
Unfortunately, it is not solvable at the moment.
To clarify, there are no obligations for anyone to resolve.
Thanks for trying to help me.
For me every comment ends here.

closed 2024-03-06 by MattBlack85