devDucks/astroarch #60: Bluetooth

sc74, 2024-03-02:
A suggestion for installing and enable Bluetooth.

Install
sudo pacman -S bluez bluez-utils bluez-hid2hci bluedevil

Uncomment in /etc/bluetooth/main.conf
DiscoverableTimeout = 0
AlwaysPairable = true
PairableTimeout = 0
AutoEnable=true

Start the service
sudo systemctl start bluetooth

With this, I was able to connect my Bluetooth keyboard and mouse without any issues.

Good day

Stéphane

MattBlack85, 2024-03-03:
Is there a widget on the tray for the Bluetooth with just bluez?

sc74, 2024-03-03:
It's the bluedevil package that implements this in KDE

It is noted in the wiki https://wiki.archlinux.org/title/bluetooth

"Bluedevil — KDE's Bluetooth tool. If there is no Bluetooth icon visible in Dolphin and in the system tray, enable it in the system tray options or add a widget. You can configure Bluedevil and detect Bluetooth devices by clicking the icon. An interface is also available from the KDE System Settings"

[image]

MattBlack85, 2024-04-11:
done in c7d65e048c25f8708b6f74801b194965a8273dea

closed 2024-04-11 by MattBlack85