devDucks/astroarch #63: CPU temperature widget

sc74, 2024-03-04:
Would it be possible to add a widget for CPU temperature?

As a suggestion, there is this package:

plasma5-applets-thermal-monitor-git

It warns of temperature limits

As a complement, we can also add the psensor package.

In addition to the temperature, it gives the fan status, the CPU usage and free memory.

wadouk, 2024-09-22:
doesn't find it under pacman
when I install some with plasmashell gui, I have one from [olib version 0.1.6](https://store.kde.org/p/2100418/) but the sensor window selector is empty

sc74, 2024-09-22:
Since the transition to qt6 this module no longer works. I haven't looked to see if anyone recreates it

sc74, 2025-11-01:
It does not work. The RPi sensors are not recognized by KDE System Monitor. An alternative is to use the psensor software from Gnome.

closed 2025-12-15 by MattBlack85