devDucks/astroarch #65: AstroArch non riparte dopo aggiornamenti

AstroVersilia, 2024-03-05:
Buonasera Mattia,
ho reinstallato un RTC, dopo aver riavviato correttamente ho effettuato l'aggiornamento ma non c'è modo di accedere a AstroArch. 

Qualche suggerimento?

Grazie,
Paolo

MattBlack85, 2024-03-05:
Su che versione sei?

AstroVersilia, 2024-03-05:
Non saprei dirti la versione. Considera che è una nuova installazione e ho
aggiornato tutto.

MattBlack85, 2024-03-05:
Hai usato un'immagine già avevi o l'hai scaricata di recente?
Puoi spiegare meglio cosa intendi per mancato accesso?
Stai usando vnc? Carica vnc e poi dà errore? Non carica nemmeno la pagina di vnc? Hai provato ad attaccarlo a un monitor? La rete la vedi?

Lukadi80, 2024-03-06:
Buongiorno Mattia... dopo gli ultimi aggiornamenti è successo anche a me che non riesco più ad accedere sia tramite hot spot che con real vnc  via lan! È proprio successo al riavvio dopo l'aggiornamento!

AstroVersilia, 2024-03-06:
Buongiorno Mattia,

Versione 1.7 scaricata a dicembre, andata egregiamente fino a ieri.
Poi dopo aver installato il nuovo RTC e avere aggiornato ho cominciato ad
avere problemi, praticamente il sistema si avvia perché riesco tramite web
ad arrivare alle directory ma non acceso più né tramite hotspot novnc sia
con vnc.
Pensando che il problema fosse dovuto al nuovo RTC ho provato a toglierlo
ma niente da fare.
Ho ripreso la mia immagine, ho flashato, tutto ok. Installato RTC tutto ok,
rifaccio aggiornamento e fine dei giochi, non acceso più.
Tant'è che ho provato con fing a vedere se vedevo il raspi in rete e vedo
non uno bensì 3 ip.
Sono ad un punto fermo.
Ciao e grazie
Paolo

MattBlack85, 2024-03-06:
@AstroVersilia @Lukadi80 potete scaricare l'ultima versione (1.8) e provare? La 1.7 va in pensione forzata in ogni caso tra 1 settimana 🤓

AstroVersilia, 2024-03-06:
Ok provo a rifare le stesse operazioni. Ti tengo aggiornato sui risultati.
Grazie mille per il momento.
Paolo

AstroVersilia, 2024-03-06:
Ciao Mattia.

Installato, installato RTC, aggiornato.
Funziona tutto, almeno pare. 
Bella l'interfaccia grafica, stasera provo a fare i profili su kstars.
Al momento ti dico grazie, puoi chiudere l'issue. 
A presto,
Paolo