devDucks/astroarch #67: Pi5 with a 12v battery

sc74, 2024-03-06:
Hi,

I have just tested my pi5 with my 12v battery that I use outdoors and a 12v 5v 5A converter.

https://fr.aliexpress.com/item/1005002405495827.html?spm=a2g0o.order_list.order_list_main.15.540c5e5bnK3BYy&gatewayAdapt=glo2fra

The converter is supposed to give 9~24V input: 5.2V/6A/30W output

Result: The pi5 refuses to boot on the 1T SSD. It indicates that we must add usb_max_current_enable=1 in the config.txt file.

After adding this line in the file, everything works after connecting the mount, cooled camera, gps, guide camera and eaf.

A special feature, the SSD must be plugged into the first USB 3 port, the top one.

Should this option be added to AstroArch or at least noted as information?

I hope this will help users.

In image, the amperage values ​​of the USB ports depending on the devices

[image]

teoteo, 2024-03-07:
I think I have the same [buck converter](https://www.aliexpress.com/item/1005003601377255.html) but using it with the [PiMoroni NVMe Base](https://shop.pimoroni.com/products/nvme-base?variant=41219587178579) with a Lexar NM710 SSD 1TB it boots without `usb_max_current_enable=1` in /boot/config.txt. It boots with the same SSD in an Orico USB case.
On Pi OS I get a warning about the max current but boots. 
On AstroArch I get:
```
lsusb -v | grep -E "^Bus | MaxPower"
Couldn't open device, some information will be missing
Couldn't open device, some information will be missing
Couldn't open device, some information will be missing
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
    MaxPower                0mA
Bus 001 Device 002: ID 2109:2817 VIA Labs, Inc. USB2.0 Hub             
    MaxPower                0mA
Bus 001 Device 003: ID 1a40:0101 Terminus Technology Inc. Hub
    MaxPower              100mA
Bus 001 Device 004: ID 067b:23d3 Prolific Technology, Inc. USB-Serial Controller 
    MaxPower              100mA
Bus 001 Device 005: ID 03c3:290f ZWO ASI290MM Mini
Couldn't open device, some information will be missing
    MaxPower              480mA
Bus 001 Device 006: ID 1546:01a7 U-Blox AG [u-blox 7]
    MaxPower              100mA
Couldn't open device, some information will be missing
Couldn't open device, some information will be missing
Bus 001 Device 007: ID 1a86:7523 QinHeng Electronics CH340 serial converter
    MaxPower               96mA
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
    MaxPower                0mA
Couldn't open device, some information will be missing
Bus 002 Device 002: ID 2109:0817 VIA Labs, Inc. USB3.0 Hub             
    MaxPower                0mA
Bus 002 Device 003: ID 0547:12b7 Anchor Chips, Inc. USB3.0 Camera
    MaxPower              800mA
Couldn't open device, some information will be missing
Couldn't open device, some information will be missing
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
    MaxPower                0mA
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
    MaxPower                0mA
``` 
Using an oscilloscope I can see before booting an handshaking phase to test the max power it can get from the power supply and sometimes the voltage reaches 4.9V.

sc74, 2024-03-08:
Yes, it's the same converter. Maybe having an NVMe instead of the USB SSD would be better. I think my problem comes more from the fact that I am using a cable longer than 1m50 to power the converter. I'm going to test with a shorter cable, but after that it will cause me other issues with my three setups too close to each other.
Afterward, I can also leave it like that, because I just tested it this evening and no issues to note. All hardware worked.

teoteo, 2024-03-10:
Now I’m waiting [this one](https://www.aliexpress.com/item/1005005301470472.html) (LTC3780 DC-DC 5-32V to 1V-30V 10A) for another project but if it will fork for Pi 5 without alerts, I’ll use it for Pi and to power the USB hub.

sc74, 2024-03-10:
It looks good. I will wait for your feedback on this converter to order it myself.

teoteo, 2024-03-16:
We will have to wait: I just received this instead of the stepdown 😤
[image]

teoteo, 2024-03-28:
> It looks good. I will wait for your feedback on this converter to order it myself.

Finally delivered.
It has a stable tension during boot
https://github.com/devDucks/astroarch/assets/789650/3daeb6ee-4565-41a2-9ff3-577fabdf76cc

but the usual alert appears on Pi OS
[image]

I think it is related to a missing USB-C [PD handshake message](https://forums.raspberrypi.com/viewtopic.php?t=358576#p2150306)

The output with a 65W PD power supply is:
[image]

With this new 10A buck converter calibrated to about 5.2V:
[image]

AlbertBirtash, 2024-04-09:
I've ran into the same problem. I also assume it has something to do with the USB PD handshake. If the Pi cannot do this, it assumes it is connected to a power supply that is not able to deliver 5A. I haven't tested it yet but maybe by setting PSU_MAX_CURRENT to 5000 might solve the problem:
https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#PSU_MAX_CURRENT

Or do you know about buck converters that are able to do USB PD?

AlbertBirtash, 2024-04-10:
I now tried this with the PSU_MAX_CURRENT option and it seems to work. There is no alert anymore. I'm now switching to another buck converter capable to provide 7A just to be sure and hopefully this finally solves the power issues.

BenoitBotton, 2024-04-20:
Very interesting thread.
I think I am facing similar issues

@sc74 
> It indicates that we must add usb_max_current_enable=1 in the config.txt file.
Where did you find this 'indication'? I am guessing in a log on a subsequent boot but so far fail to find anything on my pi (I don't know the different logs very well)

sc74, 2024-04-21:
@BenoitBotton In the documentation under Differences on Raspberry Pi 5
 https://www.raspberrypi.com/documentation/computers/raspberry-pi.html

FireNWater, 2024-06-22:
Are you folks powering your Rpi-5 thru the USB-C power port, or directly to the +5VDC GPIO pins?  I found that powering the Rpi-5 directly to the GPIO pins solves many power issues.

too-many-projects, 2025-01-02:
I don't have a Raspberry Pi 5 but I do use a Raspberry Pi 3B+ on 12 volts. I also needed a relatively high current at 5 volts and went looking for a neat, relatively cheap and reliable solution. 
What I ended up using is a US$12 in-vehicle type USB charger with a tiny bit of hackery.
This one [https://www.aliexpress.com/item/1005006054781698.html](https://www.aliexpress.com/item/1005006054781698.html).
[image]
 It does 3 amps out of each USB port but there are other types that can provide much higher currents. I cut off the cigarette lighter plug and retained the 3 port rear seat section and the cable. The cable is just two wires as it just carries 12 volts to the rear seat part of the device.  I added a plug to the end of the cable to suit my setup (D-TAP connectors). Now I have a 5 volts supply for powering my Raspberry Pi and a couple of extra USB ports to power other devices or charge my phone.
The nice thing is these chargers are designed to work on both 12 volt and 24 volt vehicles and so have a broad input voltage range. They should also cope with the voltage dropping below 12 volts as a vehicle 12 volts supply is not a very stable voltage. I just tested and the PD port was still outputting Super Speed charging into my Samsung phone while the charger was being fed 6.8 volts. So it must be a buck/boost circuit not just a buck circuit.
And then the icing on the cake is the red led voltage display, which is quite faint in daylight. So much so that I did read some complaints that the display was too faint. But that is perfect for use at night around a telescope. If it is still too bright then some appropriate tape over the display can reduce it to suit. Knowing the voltage is handy, especially if you are running portable off of a lead acid or gel cell setup.
The only thing I haven't checked is if there is any reverse polarity protection, so be careful with your wiring and use polarised connectors to avoid any mishaps.

closed 2025-10-30 by MattBlack85