devDucks/astroarch #68: update-astroach command fails on fresh Astroarch install

Astrostecco, 2024-03-09:
Hi,

I am encountering this quite weird issue where the command update-astroarch won't work, stopping after 30 seconds with the following:
error: unresolvable package conflicts detected
error: failed to prepare transaction (conflicting dependencies)

I say it's "weird" because only 2 days ago I did a fresh install on the same SD card, on the same RPi4, and everything worked fine!
After that, I formatted the SD card and installed Astroberry on it (I needed to try some trouble shooting), and today when I re-installed Astroarch it gave me these errors. I have repeated these steps twice, and for both times the same error appeared.

Attached a screenshot of the terminal.

Any idea what's happening and how to resolve?

Thanks!
Stefano

[image]

MattBlack85, 2024-03-09:
Hi Stefano, I am preparing indi for the wireplumber package as media-session is gonna be decommissioned. I tried at mine but got no issues so I released it, can you try  the following?

sudo pacman -R kstars indi-3rdparty-libs indi-3rdparty-drivers libindi pipewire-media-session

and then

sudo pacman -S kstars indi-3rdparty-drivers indi-3rdparty-libs

Astrostecco, 2024-03-09:
Hi Mattia,

sure, I am now re-flashing for the third time on a freshly formatted SD card to see if I get the same error.
Do you want me to try your commands before the update-astroarch command, or after?

MattBlack85, 2024-03-09:
before, you don't need to re-flash, the state is a good one

Astrostecco, 2024-03-09:
I tried the first command I got this:

sudo pacman -R kstars indi-3rdparty-libs indi-3rdparty-drivers libindi pipewire-media-session
[sudo] password for astronaut: 
checking dependencies...
error: failed to prepare transaction (could not satisfy dependencies)
:: removing pipewire-media-session breaks dependency 'pipewire-session-manager' required by kwin
:: removing libindi breaks dependency 'libindi' required by phd2
:: removing pipewire-media-session breaks dependency 'pipewire-session-manager' required by pipewire-jack

I am now trying the second command

Astrostecco, 2024-03-09:
This is the result after second command, shall I do/try anything else?

~ » sudo pacman -S kstars indi-3rdparty-drivers indi-3rdparty-libs                    1 ↵ astronaut@astroarch
warning: kstars-3.6.9-1 is up to date -- reinstalling
warning: indi-3rdparty-drivers-2.0.6-1 is up to date -- reinstalling
warning: indi-3rdparty-libs-2.0.6-1 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Packages (3) indi-3rdparty-drivers-2.0.6-1  indi-3rdparty-libs-2.0.6-1  kstars-3.6.9-1

Total Installed Size:  314.45 MiB
Net Upgrade Size:        0.00 MiB

:: Proceed with installation? [Y/n] y
:: Retrieving packages...
 indi-3rdparty-drivers-2.0.6-1-aa...  1270.8 KiB  1535 KiB/s 00:01 [------------------------------------] 100%
 indi-3rdparty-libs-2.0.6-1-aarch64     20.5 MiB  2.85 MiB/s 00:07 [------------------------------------] 100%
 kstars-3.6.9-1-aarch64                 81.3 MiB  4.40 MiB/s 00:18 [------------------------------------] 100%
(3/3) checking keys in keyring                                     [------------------------------------] 100%
(3/3) checking package integrity                                   [------------------------------------] 100%
(3/3) loading package files                                        [------------------------------------] 100%
(3/3) checking for file conflicts                                  [------------------------------------] 100%
(3/3) checking available disk space                                [------------------------------------] 100%
:: Processing package changes...
(1/3) reinstalling kstars                                          [------------------------------------] 100%
(2/3) reinstalling indi-3rdparty-libs                              [------------------------------------] 100%
(3/3) reinstalling indi-3rdparty-drivers                           [------------------------------------] 100%
:: Running post-transaction hooks...
(1/6) Reloading device manager configuration...
(2/6) Arming ConditionNeedsUpdate...
(3/6) Updating linux initcpios...
==> Building image from preset: /etc/mkinitcpio.d/linux-rpi.preset: 'default'
==> Using configuration file: '/etc/mkinitcpio.conf'
  -> -k 6.6.18-1-rpi -c /etc/mkinitcpio.conf -g /boot/initramfs-linux.img
==> Starting build: '6.6.18-1-rpi'
  -> Running build hook: [base]
  -> Running build hook: [udev]
  -> Running build hook: [autodetect]
  -> Running build hook: [modconf]
  -> Running build hook: [kms]
  -> Running build hook: [keyboard]
  -> Running build hook: [keymap]
  -> Running build hook: [consolefont]
==> WARNING: consolefont: no font found in configuration
  -> Running build hook: [block]
  -> Running build hook: [filesystems]
  -> Running build hook: [fsck]
==> Generating module dependencies
==> Creating gzip-compressed initcpio image: '/boot/initramfs-linux.img'
==> WARNING: errors were encountered during the build. The image may not be complete.
error: command failed to execute correctly
(4/6) Refreshing PackageKit...
(5/6) Updating icon theme caches...
(6/6) Updating the desktop file MIME type cache...

MattBlack85, 2024-03-09:
ok I went back to the old version 2.0.6-1 of libindi - will probably script the update under `update-astroarch` and for now it won't bother you anymore. If you run now update-astroarch it should show no issues

Astrostecco, 2024-03-09:
This time it worked, it downloaded all 256 upgrades and it looks like what I had two days ago.
There were some warnings which I don't remember if were there last time, so I am pasting below the outcome from the terminal just FYI.
Thank you very much for your fast feedback!! :)

:: Running post-transaction hooks...
( 1/23) Creating system user accounts...
Creating group 'geoclue' with GID 963.
Creating user 'geoclue' (Geoinformation service) with UID 963 and GID 963.
Creating group 'rtkit' with GID 133.
Creating user 'rtkit' (RealtimeKit) with UID 133 and GID 133.
( 2/23) Updating journal message catalog...
( 3/23) Reloading system manager configuration...
( 4/23) Reloading user manager configuration...
( 5/23) Updating udev hardware database...
( 6/23) Applying kernel sysctl settings...
( 7/23) Creating temporary files...
( 8/23) Reloading device manager configuration...
( 9/23) Arming ConditionNeedsUpdate...
(10/23) Updating the MIME type database...
(11/23) Updating module dependencies...
(12/23) Updating linux initcpios...
==> Building image from preset: /etc/mkinitcpio.d/linux-rpi.preset: 'default'
==> Using configuration file: '/etc/mkinitcpio.conf'
  -> -k 6.6.21-1-rpi -c /etc/mkinitcpio.conf -g /boot/initramfs-linux.img
==> Starting build: '6.6.21-1-rpi'
  -> Running build hook: [base]
  -> Running build hook: [udev]
  -> Running build hook: [autodetect]
  -> Running build hook: [microcode]
==> WARNING: architecture 'aarch64' not supported, skipping hook
  -> Running build hook: [modconf]
  -> Running build hook: [kms]
  -> Running build hook: [keyboard]
  -> Running build hook: [keymap]
  -> Running build hook: [consolefont]
==> WARNING: consolefont: no font found in configuration
  -> Running build hook: [block]
  -> Running build hook: [filesystems]
  -> Running build hook: [fsck]
==> Generating module dependencies
==> Creating gzip-compressed initcpio image: '/boot/initramfs-linux.img'
bsdtar: Failed to set default locale
bsdtar: Failed to set default locale
==> WARNING: errors were encountered during the build. The image may not be complete.
error: command failed to execute correctly
(13/23) Refreshing PackageKit...
(14/23) Updating the appstream cache...
✔ Metadata cache was updated successfully.
(15/23) Reloading system bus configuration...
(16/23) Updating fontconfig cache...
(17/23) Probing GDK-Pixbuf loader modules...
(18/23) Updating GIO module cache...
(19/23) Updating icon theme caches...
(20/23) Updating the info directory file...
(21/23) Updating the desktop file MIME type cache...
(22/23) Updating the vlc plugin cache...
(23/23) Updating X fontdir indices...

Astrostecco, 2024-03-09:
OK, I wrote too early...
After the update command I simply rebooted and this came page came out: it doesn't matter what I do, even if I input the password "astro" it will just shut the session and restart with the same login page.
[image]

MattBlack85, 2024-03-09:
@Astrostecco out of curiosity, were there any kde6/plasma 6 updates?

MattBlack85, 2024-03-09:
I think this is related to the massive change in plasma (5 to 6) happening now

Can you ssh into the machine and try `sudo pacman -S qt5-declarative` and a reboot?

Astrostecco, 2024-03-09:
> @Astrostecco out of curiosity, were there any kde6/plasma 6 updates?

I actually don't know what that means ;)

> I think this is related to the massive change in plasma (5 to 6) happening now
> 
> Can you ssh into the machine and try `sudo pacman -S qt5-declarative` and a reboot?

I did, but the same login page was coming up after a reboot. However, if I selected the "Plasma(X11)" option in the upper left menu, then it lets me log in the system and I see everything as normal (see screenshot).
[image]

MattBlack85, 2024-03-09:
gimme few mins, I am upgrading on my pi as well :) I think sddm must be updated as well (it didn't cause it was a pinned package). Will report back soon

MattBlack85, 2024-03-09:
@Astrostecco if you are on the system, I just pulled a fix for the greeter config, running `update-astroarch` again should pull the fix and on the next reboot should autologin the user

Astrostecco, 2024-03-09:
Yep, solved :)
Thanks!!!

MattBlack85, 2024-03-09:
thanks for the early report @Astrostecco, that allowed to look fast at the problem and find a possible patch; and sorry for the troubles! 

Luckily we have been fast at patching so that other users will be able to upgrade smoothly!

closed 2024-03-10 by MattBlack85