devDucks/astroarch #70: dopo aggiornamento non si accede

AstroVersilia, 2024-03-15:
Ciao Mattia

ho aggiornato e questo è il risultato

[image]

l'accesso lo faccio via novnc ma quando entro si resetta e risiamo da capo.

Cosa posso fare?

Ciao e grazie

Paolo

MattBlack85, 2024-03-15:
ciao Paolo hai aggiornato con `pacman` o con `update-astroarch`? La versione che montavi era beta o quella del link che c'è ora?

AstroVersilia, 2024-03-15:
Ciao Mattia

Ho aggiornato mediante l'icona in basso a destra che segnala gli
aggiornamenti.
La versione è quella che ho scaricato qualche giorno fa (1.8), avevo
aggiornato e andava benissimo, adesso invece dopo 500 mb di download
presenta questo problema.
C'è modo di uscirne senza dover reinstallare il sistema?

Grazie,

Paolo

MattBlack85, 2024-03-15:
@AstroVersilia dai `update-astroarch` e fai un hard reboot. Dovrebbe sistemarsi

C'è un fix di una configurazione ma viene sistemato solo con `update-astroarch`

MattBlack85, 2024-03-15:
@AstroVersilia scusa, non sono stato preciso al 100%. Intendevo dando il comando accedendo con ssh, riesci ad accedere tramite ssh?

AstroVersilia, 2024-03-15:
Purtroppo non sono così smanettone, saresti così gentile da indicarmi la
procedura?
Grazie mille

MattBlack85, 2024-03-15:
@AstroVersilia no problem, forse allora facciamo prima così:

quando ti si presenta la schermata c'è una scritta `Desktop session: plasma wayland` dovresti poter scegliere x11 lì, dopo che cambi la session in x11 metti la password per astronaut e ti parte come al solito.

Appena dentro apri il terminale e lancia `update-astroarch` e fai un hard reboot del pi. Dovrebbe sistemarsi automaticamente dopo

AstroVersilia, 2024-03-15:
Proviamo, ti tengo aggiornato

Grazie mille per ora

AstroVersilia, 2024-03-15:
Ripartito e tutto a posto.

Grazie per i sempre preziosi aiuti

A presto

Paolo

MattBlack85, 2024-03-16:
Non c'è di che

closed 2024-03-16 by MattBlack85