devDucks/astroarch #71: Raspberry Pi High Quality Camera on AstroArch

Olli781, 2024-03-17:
Hello everyone,
I have successfully installed AstroArch on my Raspberry Pi 5 and everything is working well so far. Now I would like to use the Raspberry Pi High Quality Camera. Unfortunately this is not recognized in Kstars/EKOS. Also the command

_libcamera-hello --list-cameras_

does not show any camera. Does anyone have an idea how to activate the camera?

Tanks very much, Oliver

MattBlack85, 2024-03-18:
hi @Olli781 I am not sure about the problem here, I don't have a pi camera myself but I am willing to help you troubleshoot.

The package on astroarch is pretty old and I would start from compiling the latest version, after that you can upgrade and we can check if it was just that, if not we will check further.

In the meanwhile, can you report what `v4l2-ctl --list-devices` command says?

Olli781, 2024-03-19:
Hello Mattia,
First of all, thank you very much for your help. As suggested, I updated all packages. Unfortunately the camera is still not working. Attached is the result of the command:
_v4l2-ctl --list-devices_

astronaut@astroarch
pispbe (platform:1000880000.pisp_be):
        /dev/video20
        /dev/video21
        /dev/video22
        /dev/video23
        /dev/video24
        /dev/video25
        /dev/video26
        /dev/video27
        /dev/video28
        /dev/video29
        /dev/video30
        /dev/video31
        /dev/video32
        /dev/video33
        /dev/video34
        /dev/video35
        /dev/video36
        /dev/video37
        /dev/media1
        /dev/media2

rp1-cfe (platform:1f00128000.csi):
        /dev/video0
        /dev/video1
        /dev/video2
        /dev/video3
        /dev/video4
        /dev/video5
        /dev/video6
        /dev/video7
        /dev/media0

It seems that the camera is recognized by Raspberry PI on the CSI port. Unfortunately, the "libcamera-hello" command doesn't work either.

Attached is an excerpt from my config.txt. I have added the passage in bold:

**********************************************************************************************************************
#For more options and information see:
#http://rptl.io/configtxt

initramfs initramfs-linux.img followkernel

#Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

#Additional overlays and parameters are documented
#/boot/firmware/overlays/README

#Automatically load overlays for detected cameras
camera_auto_detect=1

#Automatically load overlays for detected DSI displays
display_auto_detect=1

#Enable DRM VC4 V3D driver
dtoverlay=vc4-fkms-v3d
max_framebuffers=2
framebuffer_depth=24

#Uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

#Uncomment if you want to disable wifi or bluetooth respectively
#dtoverlay=disable-wifi
#dtoverlay=disable-bt

#Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

#Run as fast as firmware / board allows
arm_boost=1

[cm4]
#Enable host mode on the 2711 built-in XHCI USB controller.
#This line should be removed if the legacy DWC2 controller is required
#(e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]
dtparam=i2c_arm=on
dtparam=audio=on
disable_overscan=1
gpu_mem=256
**start_file=start_x.elf
fixup_file=fixup_x.dat
initramfs initramfs-linux.img followkernel
cma_lwm= cma_hwm= cma_offline_start=**
disable_splash=1
3dtparam=krnbt=on
hdmi_drive=2
dtoverlay=i2c-rtc
[pi5]
dtparam=rtc_bbat_vchg=3000000
*********************************************************************************************************

Best regards, Oliver

MattBlack85, 2024-03-19:
hi again @Olli781 

I cracked the compilation and it seems my trick works, can you try to run the following commands and report back if the camera is recognized? Please reboot the pi after before calling the camera

```bash
wget http://astromatto.com:9000/aarch64/libcamera-0.2.0-2-aarch64.pkg.tar.xz
wget http://astromatto.com:9000/aarch64/libcamera-docs-0.2.0-2-aarch64.pkg.tar.xz
wget http://astromatto.com:9000/aarch64/libcamera-ipa-0.2.0-2-aarch64.pkg.tar.xz
wget http://astromatto.com:9000/aarch64/libcamera-tools-0.2.0-2-aarch64.pkg.tar.xz
wget http://astromatto.com:9000/aarch64/rpicam-apps-1.4.3-1-aarch64.pkg.tar.xz
wget http://astromatto.com:9000/aarch64/gst-plugin-libcamera-0.2.0-2-aarch64.pkg.tar.xz
sudo pacman -U gst-plugin-libcamera-0.2.0-2-aarch64.pkg.tar.xz libcamera-0.2.0-2-aarch64.pkg.tar.xz libcamera-docs-0.2.0-2-aarch64.pkg.tar.xz libcamera-ipa-0.2.0-2-aarch64.pkg.tar.xz libcamera-tools-0.2.0-2-aarch64.pkg.tar.xz rpicam-apps-1.4.3-1-aarch64.pkg.tar.xz
```

Please note the the package rpicams-apps provides both `libcamera-*` and `rpicam-*` as valid binaries

MattBlack85, 2024-03-19:
Oh, seems there are some duplicated files around, please as a quick test remove phd2 and retry that command, to remove it `sudo pacman -R phd2` if you don't use it at all you can live without it.

Will try to fix the libcamera packages to avoid conflicts in the next days

Let me know if that works @Olli781

Olli781, 2024-03-20:
Hello Matt, we are a small step further. The libcamera-hello command works, but a library is still missing:

_~ » libcamera-hello
libcamera-hello: error while loading shared libraries: libpisp.so.1: cannot open shared object file: No such file or directory_

Any idea how you can reinstall this?

MattBlack85, 2024-03-20:
@Olli781 forgive me, I installed that yesterday but forgot to add it to the commands, to install it

`sudo pacman -Sy && sudo pacman -S libpisp`

Olli781, 2024-03-21:
Hi Matt, first of all I have to apologize for the late reply. The command "libcamera-hello" works:

_Mode selection for 1640:1232:12:P
    SRGGB10_CSI2P,640x480/0 - Score: 4504.81
    SRGGB10_CSI2P,1640x1232/0 - Score: 1000
    SRGGB10_CSI2P,1920x1080/0 - Score: 1541.48
    SRGGB10_CSI2P,3280x2464/0 - Score: 1718
    SRGGB8,640x480/0 - Score: 5504.81
    SRGGB8,1640x1232/0 - Score: 2000
    SRGGB8,1920x1080/0 - Score: 2541.48
    SRGGB8,3280x2464/0 - Score: 2718
Stream configuration adjusted
[0:12:24.939976720] [1267]  INFO Camera camera.cpp:1183 configuring streams: (0)
1640x1232-YUV420 (1) 1640x1232-BGGR16_PISP_COMP1
[0:12:24.940103590] [1268]  INFO RPI pisp.cpp:1405 Sensor: /base/axi/pcie@120000
/rp1/i2c@80000/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Sele  cted CFE format: 1640x1232-PC1B
dmaHeap allocation failure for rpicam-apps0
ERROR: *** failed to allocate capture buffers for stream ***_

Only the preview doesn't work, but that's not a bad thing. The camera is not yet recognized under kstars. I think for this you still have to install the "indi pylibcamera - driver":

_https://github.com/scriptorron/indi_pylibcamera_

Is it possible under astroarch?

MattBlack85, 2024-03-21:
Hi @Olli781 great success! Indi pylibcamera is already available, you can try to install and have a go with `sudo pacman -S indi_pylibcamera`

I am not sure it works, this was compiled a long time ago, if you can test is tho and report back it is not working I will recompile a more recent version tomorrow 🙂

MattBlack85, 2024-03-22:
hi @Olli781 if you installed the package `indi_pylibcamera` I suggest to remove it using `sudo pacman -R indi_pylibcamera` and install the newest version with `sudo pacman -Sy indi-pylibcamera`

I recompiled a few packages and I added also the dev rule to allow `video` users to use the pi allocator (without sudo I couldn't get indi pylibcamera to work). indi pylibcamera seems to work(ish) although I am using a normal camera and it's failing with some weird error, maybe it will work with a pi camera?

Olli781, 2024-03-22:
Hey Matt,
it partially works. If I use the command "indiserver -v indi_pylibcamera" the indi driver starts:       

_2024-03-22T18:59:58: startup: indiserver -v indi_pylibcamera
2024-03-22T18:59:58: Driver indi_pylibcamera: pid=1395 rfd=6 wfd=6 efd=7
2024-03-22T18:59:58: listening to port 7624 on fd 5
2024-03-22T18:59:58: Local server: listening on local domain at: @/tmp/indiserve                                  r
2024-03-22T18:59:59: Driver indi_pylibcamera: [0:03:54.891807630] [1395] ERROR I                      PAModule ipa_module.cpp:172 Symbol ipaModuleInfo not found
2024-03-22T18:59:59: Driver indi_pylibcamera: [0:03:54.891831316] [1395] ERROR I                     PAModule ipa_module.cpp:292 v4l2-compat.so: IPA module has no valid info
2024-03-22T18:59:59: Driver indi_pylibcamera: [0:03:54.891859318] [1395]  INFO C                               amera camera_manager.cpp:284 libcamera v0.2.0
2024-03-22T18:59:59: Driver indi_pylibcamera: [0:03:54.902340682] [1397]  INFO R                       PI pisp.cpp:662 libpisp version v1.0.4 2e1ad45c8f0e-dirty 19-03-2024 (12:49:28)
2024-03-22T18:59:59: Driver indi_pylibcamera: [0:03:54.914271682] [1397]  INFO R                                PI pisp.cpp:1121 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx219@10                      to CFE device /dev/media2 and ISP device /dev/media0 using PiSP variant BCM2712                       _C0
2024-03-22T18:59:59: Driver indi_pylibcamera: indi_pylibcamera.indidevice-INFO-                      found cameras: [{'Model': 'imx219', 'Location': 2, 'Rotation': 180, 'Id': '/base                        /axi/pcie@120000/rp1/i2c@80000/imx219@10', 'Num': 0}]
2024-03-22T18:59:59: Driver indi_pylibcamera: indi_pylibcamera.indidevice-INFO-selected logging level: LOGGING_INFO_

Unter KSTARS the pylibcamera is also available. But when I chose the pylibcamera, KSTARS crashed.
I'm trying to completely reinstall Astroach and see if it works. We are definitely getting closer to our goal :-)

MattBlack85, 2024-03-22:
I believe is the same error I am seeing then. Can you kindly turn on CCD logging and paste me the error?

For me it was python crashing because a prop INDI_RAW doesn't exist

Olli781, 2024-03-23:
Hi Matt,
I tried another way: I started the Indi driver on the PI5 and then connected to the PI5 via my Windows computer and KSTARS.
A connection is also established there:

_2024-03-23T08:51:40: [INFO] configuration /home/astronaut/.indi_pylibcamera/CONFIG1.json does not exist 
2024-03-23T08:51:39: [INFO] camera properties: {'Model': 'imx219', 'UnitCellSize': (1120, 1120), 'ColorFilterArrangement': 0, 'Location': 2, 'Rotation': 180, 'PixelArraySize': (3280, 2464), 'PixelArrayActiveAreas': [(8, 8, 3280, 2464)], 'ScalerCropMaximum': (0, 0, 0, 0), 'SystemDevices': (20738, 20740, 20742, 20744, 20746, 20748, 20750, 20741, 20743, 20745, 20747, 20749)} 
2024-03-23T08:51:39: [INFO] opening camera 
2024-03-23T08:51:39: [INFO] closing camera 
2024-03-23T08:51:39: [INFO] connecting to camera /base/axi/pcie@120000/rp1/i2c@80000/imx219@10_ 

But when I want to take a picture, it doesn't work. Here is an excerpt from PUTTY:

_~ » indiserver -v indi_pylibcamera                          astronaut@astroarch
2024-03-23T08:44:17: startup: indiserver -v indi_pylibcamera
2024-03-23T08:44:17: Driver indi_pylibcamera: pid=1182 rfd=6 wfd=6 efd=7
2024-03-23T08:44:17: listening to port 7624 on fd 5
2024-03-23T08:44:17: Local server: listening on local domain at: @/tmp/indiserve                                                    r
2024-03-23T08:44:19: Driver indi_pylibcamera: [0:00:59.297232535] [1182] ERROR I                                                    PAModule ipa_module.cpp:172 Symbol ipaModuleInfo not found
2024-03-23T08:44:19: Driver indi_pylibcamera: [0:00:59.297265554] [1182] ERROR I                                                    PAModule ipa_module.cpp:292 v4l2-compat.so: IPA module has no valid info
2024-03-23T08:44:19: Driver indi_pylibcamera: [0:00:59.297304721] [1182]  INFO C                                                    amera camera_manager.cpp:284 libcamera v0.2.0
2024-03-23T08:44:19: Driver indi_pylibcamera: [0:00:59.314660219] [1184]  INFO R                                                    PI pisp.cpp:662 libpisp version v1.0.4 2e1ad45c8f0e-dirty 19-03-2024 (12:49:28)
2024-03-23T08:44:19: Driver indi_pylibcamera: [0:00:59.343032495] [1184]  INFO R                                                    PI pisp.cpp:1121 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx219@10                                                     to CFE device /dev/media0 and ISP device /dev/media1 using PiSP variant BCM2712                                                    _C0
2024-03-23T08:44:19: Driver indi_pylibcamera: indi_pylibcamera.indidevice-INFO-                                                     found cameras: [{'Model': 'imx219', 'Location': 2, 'Rotation': 180, 'Id': '/base                                                    /axi/pcie@120000/rp1/i2c@80000/imx219@10', 'Num': 0}]
2024-03-23T08:44:19: Driver indi_pylibcamera: indi_pylibcamera.indidevice-INFO-                                                     selected logging level: LOGGING_INFO
2024-03-23T08:44:53: Client 9: new arrival from 192.168.178.52:52396 - welcome!
2024-03-23T08:44:53: Driver indi_pylibcamera: indi_pylibcamera.indidevice-INFO-                                                     connecting to camera /base/axi/pcie@120000/rp1/i2c@80000/imx219@10
2024-03-23T08:44:53: Driver indi_pylibcamera: indi_pylibcamera.indidevice-INFO-                                                     closing camera
2024-03-23T08:44:53: Driver indi_pylibcamera: indi_pylibcamera.indidevice-INFO-                                                     opening camera
2024-03-23T08:44:53: Driver indi_pylibcamera: [0:01:33.027540807] [1182] ERROR I                                                    PAModule ipa_module.cpp:172 Symbol ipaModuleInfo not found
2024-03-23T08:44:53: Driver indi_pylibcamera: [0:01:33.027561955] [1182] ERROR I                                                    PAModule ipa_module.cpp:292 v4l2-compat.so: IPA module has no valid info
2024-03-23T08:44:53: Driver indi_pylibcamera: [0:01:33.027591622] [1182]  INFO C                                                    amera camera_manager.cpp:284 libcamera v0.2.0
2024-03-23T08:44:53: Driver indi_pylibcamera: [0:01:33.042728139] [1188]  INFO R                                                    PI pisp.cpp:662 libpisp version v1.0.4 2e1ad45c8f0e-dirty 19-03-2024 (12:49:28)
2024-03-23T08:44:53: Driver indi_pylibcamera: [0:01:33.055101805] [1188]  INFO R                                                    PI pisp.cpp:1121 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx219@10                                                     to CFE device /dev/media0 and ISP device /dev/media1 using PiSP variant BCM2712                                                    _C0
2024-03-23T08:44:53: Driver indi_pylibcamera: [0:01:33.056552082] [1182] ERROR I                                                    PAModule ipa_module.cpp:172 Symbol ipaModuleInfo not found
2024-03-23T08:44:53: Driver indi_pylibcamera: [0:01:33.056572082] [1182] ERROR I                                                    PAModule ipa_module.cpp:292 v4l2-compat.so: IPA module has no valid info
2024-03-23T08:44:53: Driver indi_pylibcamera: [0:01:33.056597675] [1182]  INFO C                                                    amera camera_manager.cpp:284 libcamera v0.2.0
[... cut ...]
2024-03-23T08:46:04: Driver indi_pylibcamera:               ^^^^^^^^^^^^^^^^^^^^                                                    ^^^^^^^
2024-03-23T08:46:04: Driver indi_pylibcamera: FileNotFoundError: [Errno 2] No su                                                    ch file or directory: '/usr/lib/python3.11/site-packages/astropy_iers_data/data/                                                    eopc04.1962-now'_

What exactly do you mean by "CCD logging"? If you briefly describe what you mean, I can send you the error report.

By the way, thank you for your great support!

MattBlack85, 2024-03-23:
hi @Olli781! Thank you very much for trying again, I believe yesterday crashes were due to the udev rules not triggering when you installed pylibcamera (my fault :| ) a reboot likely fixed it.

When you start ekos, there is a series of icons (where also the indi one is) and there should be a button called "Logs", if you click there you can enable/disable logging and the logging level for specific components (and decide if you want to log to file), if you choose verbose logging to file and mark the CCD, the camera logs will flow to the logfile Kstars makes (Kstars logs files are found under `~/.local/share/kstars/logs`).

Anyway, it seems starting indiserver by hand also logs all python errors, so we are fine here :)

It seems to me that astropy does some magic stuff and is trying to download some files but it's not able to do so, it seems those URLs are working, is your Pi connected to the internet?

MattBlack85, 2024-03-23:
or maybe packaged astropy is missing some files, I will check that

MattBlack85, 2024-03-23:
ah yes, this seems indeed to be an error directly related to the astropy-iers-data package, give a while to figure it out

MattBlack85, 2024-03-23:
@Olli781 I packaged a corrected version of astropy iers data, if you update the system with `update-astroarch` it should pull the newest version which should fix your issue. Let me know!

Olli781, 2024-03-25:
Hi Matt,
it works!!!
The camera is recognized correctly and the driver runs stable. Here is the log file:

_[2024-03-24T22:17:36.844 UTC INFO ][   org.kde.kstars.ekos.capture] - "Capture exposure = 1 sec, type = Light, filter = , upload mode = 0, batch mode = false, seq prefix = _%t_%T_%F_%t_%T_%F_Light, format = RAW, encoding = , binning = 1x1, Cannot subframe"
[2024-03-24T22:17:36.847 UTC INFO ][   org.kde.kstars.ekos.capture] - "Capturing 1.000-second  image..."
[2024-03-24T22:17:36.892 UTC INFO ][           org.kde.kstars.indi] - indi_pylibcamera :  "[INFO] exposure settings: CameraSettings: FastExposure=False, DoRaw=True, ProcSize=(3280, 2464), RawMode={'label': '3280x2464 RGGB 10bit', 'size': (3280, 2464), 'true_size': (3280, 2464), 'camera_format': 'SRGGB10', 'bit_depth': 10, 'binning': (1, 1)}, CameraControls={'ExposureTime': 1000000, 'AnalogueGain': 10.666666984558105, 'AeEnable': False, 'AeConstraintMode': <AeConstraintModeEnum.Normal: 0>, 'AeExposureMode': <AeExposureModeEnum.Normal: 0>, 'AeMeteringMode': <AeMeteringModeEnum.CentreWeighted: 0>, 'AwbEnable': False, 'AwbMode': <AwbModeEnum.Auto: 0>, 'Brightness': 0.0, 'ColourGains': (2.0, 2.0), 'Contrast': 1.0, 'ExposureValue': 0.0, 'NoiseReductionMode': <NoiseReductionModeEnum.Off: 0>, 'Saturation': 1.0, 'Sharpness': 0.0} "
[2024-03-24T22:17:36.893 UTC INFO ][           org.kde.kstars.indi] - indi_pylibcamera :  "[INFO] reconfiguring camera "
[2024-03-24T22:17:38.087 UTC INFO ][           org.kde.kstars.indi] - indi_pylibcamera :  "[INFO] got exposed frame "
[2024-03-24T22:17:39.396 UTC INFO ][           org.kde.kstars.indi] - indi_pylibcamera :  "[INFO] preparing frame as BLOB: 16171200 bytes "
[2024-03-24T22:17:39.396 UTC INFO ][           org.kde.kstars.indi] - indi_pylibcamera :  "[INFO] sending BLOB "
[2024-03-24T22:17:40.364 UTC INFO ][   org.kde.kstars.ekos.capture] - "Captured /tmp/kstars.fQWnNX"
[2024-03-24T22:17:40.367 UTC INFO ][   org.kde.kstars.ekos.capture] - "Download Time: 0.60 s, New Download Time Estimate: 0.60 s."
[2024-03-24T22:17:41.060 UTC WARN ][              kf.notifications] - Failed to play sound with canberra: File or data not found_

Only one warning appears when updating:
_warning: gst-plugin-libcamera: local (0.2.0-2) is newer than extra (0.2.0-1)
warning: libcamera: local (0.2.0-2) is newer than extra (0.2.0-1)
warning: libcamera-docs: local (0.2.0-2) is newer than extra (0.2.0-1)
warning: libcamera-ipa: local (0.2.0-2) is newer than extra (0.2.0-1)
warning: libcamera-tools: local (0.2.0-2) is newer than extra (0.2.0-1)
 there is nothing to do_

Next I will try to install astroarch on an SSD. That should bring a performance gain.

Thank you again for your great support. Astroarch is a really great project!

MattBlack85, 2024-03-25:
@Olli781 I am happy it finally works :D it's been fun digging into libcamera and other stuff.

The warning are fine for now, those packages are unstable - please ignore them and remember, if you make a new installation on a SSD you will need to pull the libcamera custom packages with 

``````bash
wget http://astromatto.com:9000/aarch64/libcamera-0.2.0-2-aarch64.pkg.tar.xz
wget http://astromatto.com:9000/aarch64/libcamera-docs-0.2.0-2-aarch64.pkg.tar.xz
wget http://astromatto.com:9000/aarch64/libcamera-ipa-0.2.0-2-aarch64.pkg.tar.xz
wget http://astromatto.com:9000/aarch64/libcamera-tools-0.2.0-2-aarch64.pkg.tar.xz
wget http://astromatto.com:9000/aarch64/rpicam-apps-1.4.3-1-aarch64.pkg.tar.xz
wget http://astromatto.com:9000/aarch64/gst-plugin-libcamera-0.2.0-2-aarch64.pkg.tar.xz
sudo pacman -U gst-plugin-libcamera-0.2.0-2-aarch64.pkg.tar.xz libcamera-0.2.0-2-aarch64.pkg.tar.xz libcamera-docs-0.2.0-2-aarch64.pkg.tar.xz libcamera-ipa-0.2.0-2-aarch64.pkg.tar.xz libcamera-tools-0.2.0-2-aarch64.pkg.tar.xz rpicam-apps-1.4.3-1-aarch64.pkg.tar.xz && sudo pacman -S libpisp
```

This week I will work on automating this step in a similar fashion I did for bleeding packages so it will be easy to install them and possibly they will be the official ones for raspberris in one of the next release.

I will let you know once this shortcut to install the rpi libcamera packages will be ready!

and last but not least, thanks for your support! :heart: great to hear people like this project

MattBlack85, 2024-04-11:
this should be done, in theory now, if one runs `sudo pacman -S indi-pylibcamera` it should pull libcamera for the raspberry directly

MattBlack85, 2024-04-16:
hi @ripkrizbi thanks for the report, I noticed I may have packaged a wrong version of python-v4l2 which is indeed missing that attribute. Will have a look soon

MattBlack85, 2024-04-16:
hi @ripkrizbi can you try to run `update-astroarch` and tell me if that is fixed? I shipped the newest version (0.3.4) of python-v4l2 now which bring in the failed attr `V4L2_CID_MPEG_VIDEO_H264_PROFILE`

MattBlack85, 2024-04-16:
happy it works!

MattBlack85, 2024-04-23:
think it's time to close this one as fixed :) 

feel free to re-open if I missed something

closed 2024-04-23 by MattBlack85