devDucks/astroarch #72: Update broke AstroArch

AstroMan85, 2024-03-18:
I setup AstroArch a few months ago and everything was configured for my equipment and working well. Recently did and update-astroarch now NoVNC will not work as it seems to have switched from x11 to wayland and RDP is not compatible with wayland...... Everything was working great but idk the location of the config files to make the changes back to how it was before the update? I have tried finding the config file with no luck. If you could provide where and what changes need to be made it? Thanks in advance for the assistance.

MattBlack85, 2024-03-18:
Hi @AstroMan85 if you were on 1.7 or an older version the update won't be able to fix all the issues. Version 1.8 changed some internal behaviors which I wasn't able to port as an update script.

I would suggest to install a fresh 1.8

X11vnc is also gone in favor of Tigervnc

If you still want to try, this that may work (but I have no idea if it will):
- `sudo pacman -Rcs x11vnc && sudo pacman -S tigervnc` and a hard reboot

Alternatively, you can use astromonitor (already installed but feel free to have a look here https://github.com/MattBlack85/astro_monitor) to backup your actual profiles (indi, ekos, phd2) and restore them once you install 1.8 fresh

PS: when on 1.8 update-astroarch will pull the new plasma and you need a hard reboot of the pi (from the desktop it won't work) after that

AstroMan85, 2024-03-18:
Thanks for the response, I'll do a fresh install.  Thanks for the info, helped greatly as I was going in circles and starting to band my head against the wall.

MattBlack85, 2024-03-18:
sure thing, lemme know if everything works!

closed 2024-03-21 by MattBlack85