devDucks/astroarch #74: Sesto Senso 2 AF Overscan doesn't work

Astrostecco, 2024-03-21:
Hi,

I am having a very specific issue with my Sesto Senso 2 focuser and the AF Overscan parameter.

This issue has actually been solved a few weeks ago, see https://www.indilib.org/forum/development/14350-indi-sesto-senso-2-fix-for-af-overscan-not-working-correctly.html

I am wondering if this could be fixed in AstroArch directly by using update-Astroarch, or if I have to start coding and playing with the KStars/INDI code myself?
The first would be preferable obviously ;)

Thanks!
Stefano

MattBlack85, 2024-03-21:
Hellooooooo @Astrostecco, we do not provide a "bleeding edge" version of INDI at the moment, whenever they push a new tag on github, I make a new release of the package.

As a bimonthly package, I believe the new version will be released towards the end of the next week.

What we can do? If you want, I may prepare today indi which is is built from master, just bear with me if it won't install at the first attempt as it will be a world first for me 🤓

I should be able to release it in few hours (possibly 2/3) and you may give it a go

Astrostecco, 2024-03-21:
Hi Mattia,

if a new package is going to be released in one or two weeks I can wait, I don't want you to try new things un-necessarily!!
Let me know if you want to test something anyway, if I can be of any help :)

Cheers,
Stefano

MattBlack85, 2024-03-21:
nah, no real issues :smile:  the building pipeline is in a pretty good shape and it will be easy to clone it for bleeding edge. This may also help testing if new features will work before official release, think I am gonna go for it

Astrostecco, 2024-03-21:
OK, let me know if you want me to test it, I have anyway backed-up my SD card so even in case something gets broken I can restore it ;)

teoteo, 2024-03-21:
> This may also help testing if new features will work before official release, think I am gonna go for it

Will it includes KStars bleeding edge too?

MattBlack85, 2024-03-21:
why not, bleeding from all edges :drop_of_blood:

MattBlack85, 2024-03-22:
if you update now with `update-astroarch` you will have 2 new commands

`use-astroarch-bleeding-edge` to use unstable astro software compiled from git branches and `use-astro-stable` to go back to tagged releases.

Enjoy!

closed 2024-03-22 by MattBlack85

Astrostecco, 2024-03-23:
Thank you very much Mattia, I'll try this today :)

Astrostecco, 2024-03-23:
Hi Mattia, everything is working flawlessly now, thank you very much for this fix, you have been super quick!! :)

MattBlack85, 2024-03-23:
yay! Clear skies then!