devDucks/astroarch #79: Driver Indi per Focheggiatore

zurigo66, 2024-04-23:
Salve a tutti.
Ho da poco acquistato su Aliexpress questo focheggiatore elettrico molto simile (a parte il colore) al modello EAf della ZWO. Viene definito come GEMINI Electric focuser pro. Sul sito viene dichiarato che è compatibile con driver Ascom e Indi quindi gestibile sia con Nina sia con Ekos.  Possiedo sia Astroberry che Astroarch col quale mi trovo MOLTO bene ma di questo dispositivo non ho trovato il driver corrispondente, nemmeno tra gli other, può essere che non sia stato ancora rilasciato il driver Indi? 
Mi potete aiutare? 
Grazie in anticipo!!
Michele

teoteo, 2024-04-23:
Ciao, può essere che abbiano usato un driver già presentente. Hai già provato a sceglierne altri? 
Proverei ZWO EAF, MoonLite e quelli presenti in DIY. Puoi anche chiedere al produttore qual è il driver adatto.

MattBlack85, 2024-04-23:
@zurigo66 concordo con @teoteo 

ho cercato il codice INDI ma non c'è un solo riferimento a "gemini" quindi le opzioni sono due, una bella e una brutta :D

Opzione 1 (la bella) - usa un driver di altre case (o magari uno libero come appunto il moonlight) ma non c'è nessun riferimento a esso sulla pagina quindi bisogna per forza chiedere al venditore
Opzione 2 (la brutta) - hanno scritto fesserie sull'offerta e INDI non è supportato

MattBlack85, 2024-04-23:
non sono sicuro del significato, ma il venditore dice che il driver è
```
Indi in diy-myfocuser
```

ce ne sono 4 da quello che vedo, ci sarebbe da provarli

teoteo, 2024-04-23:
> non sono sicuro del significato, ma il venditore dice che il driver è
> 
> ```
> Indi in diy-myfocuser
> ```
> 
> ce ne sono 4 da quello che vedo, ci sarebbe da provarli

È probabile che sia il [MyFocuserPro2](https://sourceforge.net/projects/arduinoascomfocuserpro2diy/) di Robert Brown.
@zurigo66 Se si connette, dovresti vedere qualcosa del genere:
[image]

zurigo66, 2024-04-23:
Allora ragazzi vi comunico con gioia che effettivamente come suggerito da teoteo si tratta proprio del driver MyFocuserPro2.  
Collegato immediatamente e funziona perfettamente :-) :-) :-)
Spero che possa essere utile ad altri utenti della comunity.
Grazie a tutti!!
P.S.  Mattia continua così sei un grande!!

MattBlack85, 2024-04-25:
grazie, qua chiudo allora se è tutto a posto!

closed 2024-04-25 by MattBlack85

zurigo66, 2024-04-25:
Tutto a posto di nuovo grazie :-)