devDucks/astroarch #80: AstrArch-1.8: When RasPi connects (W)LAN it should adopt its timezone, date and time

sophisticatedDreamer, 2024-05-03:
When RasPi connects (W)LAN it should adopt its timezone, date and time.

MattBlack85, 2024-05-10:
when connected to the network it does what reported, it may take longer times for clocks that are way off but eventually it will sync. How long you waited ?

gilbertgit, 2024-05-11:
I have the same issue. I had to resort to setting the date and time manually via the terminal. I tried using `timedatectl` but no luck, so I used `sudo date -s` to set it. Could never get it to do it automatically. Every time I turn on the pi, the date defaults to March 4th.

MattBlack85, 2024-05-11:
@gilbertgit would you send me the result of `sudo systemctl status chronyd`?

Also, what kind of network you use?

sophisticatedDreamer, 2024-05-11:
@MattBlack85 - The command `sudo systemctl status chronyd` yields this output on my **RasPi4** for me:

`
○ chronyd.service - NTP client/server
     Loaded: loaded (/usr/lib/systemd/system/chronyd.service; disabled; preset: disabled)
     Active: inactive (dead)
       Docs: man:chronyd(8)
             man:chrony.conf(5)
`

MattBlack85, 2024-05-11:
@sophisticatedDreamer ah that would explain why time doesn't sync although it is weird it is disabled, can you try to enable it back by running `sudo systemctl enable --now chronyd` amd see if it sync?

In theory it should survive a reboot but what's more strange is that the service is enabled out of the box and I am unsure why it is disabled

sophisticatedDreamer, 2024-05-11:
By the way, after re-flashing and re-launching _AstroArch_ (AA), I had to plug it to an Ethernet cable to configure the WiFi. Because it was plugged to a cable, date/time was immediately adopted from the LAN.

I had expected this same behavior when using WiFi.

sophisticatedDreamer, 2024-05-11:
@MattBlack85 - I agree ... setting a system service with `systemctl enable .... sevice` is the command to letting the service survive a reboot.

sophisticatedDreamer, 2024-05-11:
I will now shutdown AA@RasPi4 for a couple of hours and reboot it then. It should then adopt time/date immediately from WiFi as expected ...

gilbertgit, 2024-05-11:
Yeah, that was the problem. The service was disabled! I `sudo systemctl enable` and `sudo systemctl  start` and it got the datetime right away! I did a reboot of the pi and it booted back up with the correct datetime. Thank you for your help!

sophisticatedDreamer, 2024-05-11:
> I will now shutdown AA@RasPi4 for a couple of hours and reboot it then. It should then adopt time/date immediately from WiFi as expected ...

No matter what and how often I do, `sudo systemctl enable` (should survive reboot) and `sudo systemctl  start` never survive a reboot. Seems that time/date from the last shutdown are stored and reused after next reboot, even hours later, with no automatic update.

MattBlack85, 2024-05-13:
I know timesyncd renders chrony dead, maybe that is enabled? can you check kindly @sophisticatedDreamer if `sudo systemctl status systemd-timesyncd` reports the service active by any chance?

sophisticatedDreamer, 2024-05-13:
> I know timesyncd renders chrony dead, maybe that is enabled? can you check kindly @sophisticatedDreamer if `sudo systemctl status systemd-timesyncd` reports the service active by any chance?

Sure:
`sudo systemctl status systemd-timesyncd                                                                 astronaut@astroarch
○ systemd-timesyncd.service - Network Time Synchronization
     Loaded: loaded (/usr/lib/systemd/system/systemd-timesyncd.service; enabled; preset: enabled)
     Active: inactive (dead)
       Docs: man:systemd-timesyncd.service(8)`

sophisticatedDreamer, 2024-05-13:
**Off topic:** I had to edit my former post because clicking the [Code] button to format code, only sets _one_ backtick around the pasted code while _three_ backticks are required to get the right formatting.

MattBlack85, 2024-05-13:
@sophisticatedDreamer yes, i would expect timesyncd to work equally well if chrony is not active. From what i remember some people using mobile network routers were having issues with time syncing and we switched to chrony, but I am open to change to whatever works for everyone

One lastr trial could be the following:
- disable timesyncd (seems it is enabled but dead and it's weird cause of this https://github.com/devDucks/astroarch/blob/main/astroarch_build.sh#L93)
enabling chrony

check if that works

sophisticatedDreamer, 2024-05-13:
```
sudo systemctl status systemd-timesyncd                                                                                                   astronaut@astroarch
○ systemd-timesyncd.service - Network Time Synchronization
     Loaded: loaded (/usr/lib/systemd/system/systemd-timesyncd.service; enabled; preset: enabled)
     Active: inactive (dead)
       Docs: man:systemd-timesyncd.service(8)
------------------------------------------------------------------------------------------------------------------------------------------------------------------
~ » sudo systemctl status chronyd                                                                                                         4 ↵ astronaut@astroarch
● chronyd.service - NTP client/server
     Loaded: loaded (/usr/lib/systemd/system/chronyd.service; enabled; preset: disabled)
     Active: active (running) since Tue 2024-03-05 02:41:23 CET; 2 months 8 days ago
       Docs: man:chronyd(8)
             man:chrony.conf(5)
    Process: 345 ExecStart=/usr/bin/chronyd $OPTIONS (code=exited, status=0/SUCCESS)
   Main PID: 365 (chronyd)
      Tasks: 1 (limit: 8528)
        CPU: 100ms
     CGroup: /system.slice/chronyd.service
             └─365 /usr/bin/chronyd

May 13 15:16:13 astroarch chronyd[365]: System clock was stepped by 6003256.849491 seconds
May 13 15:16:13 astroarch chronyd[365]: System clock TAI offset set to 37 seconds
May 13 15:17:19 astroarch chronyd[365]: Selected source 78.46.102.180 (2.arch.pool.ntp.org)
------------------------------------------------------------------------------------------------------------------------------------------------------------------
```

MattBlack85, 2024-05-13:
the date from march is I believe the date I baked the image, wondering if a `hwclock` would move that to a more recent date

MattBlack85, 2024-06-11:
as discovered by @sc74 trying to change the time from Plasma was causing chrony to disable and additionally it wasn't even working - this should be fixed in 1.9.

Feel free to re-open should it be still broken

closed 2024-06-11 by MattBlack85