devDucks/astroarch #81: Porting to Odroid N2-N2+

Ryther, 2024-05-14:
Hi! I would like to try to port your fantastic project to my Odroid N2.
Is it possible? I've read the BUILD.md but it looks like is made for the rpi, or is it usable with any aarch64 hardware?

Regards,
Edoardo

MattBlack85, 2024-05-14:
Hi Edoardo, if you change the kernel to the Arch one and drop the config related to it it will work on any aarch64 system in theory. I was able to run it on qemu with the generic kernel and on Oracle cloud on a aarch64 cpu

This for the arch_build script. You likely need to tweak some stuff for partition and similar things for your specifoc SoC but should work

MattBlack85, 2024-05-17:
@Ryther there is a pal in discord which runs AstroArch on the lattepanda, so I believe porting to aarch64 wouldn't be that big of an issue

Ryther, 2024-05-20:
Thank you for the head up! I never did something like this, so a little help would be priceless.
Will try to contact the user on discord

MattBlack85, 2026-09-23:
hi @Ryther after almost 2 years from this we are expanding to more boards. I have successfully booted the system on an orange pi and I have an image for Odroid N2+ that could be tested, just ping me back if you'd be still interested :D

Ryther, 2026-09-24:
Hi @MattBlack85 , thank for reaching out! The board in question is not used by me anymore, it's now used as a small setup by my girlfriend. I don't have a need anymore but if I have a spare sdcard I could try it for you.