devDucks/astroarch #82: GPS support

waldiniBC, 2024-06-05:
Greetings-

I am running the latest archastro on a rpi 5. So far, so good. I am trying to get the gps working with a UBX-M8130 based usb dongle. GPSD is running and I see live values from the gps in the include xgps program. However, I don't see an option for GPSD in the INDI control panel. Do I need to install a seperate driver to make it work? Or is there a issue with the archastro distribution that is preventing gps support. 

Thanks

MattBlack85, 2024-06-05:
Hi there, the driver should be there. Let me check and I will report back.

lhanson, 2024-06-06:
For reference, I had to install the GPS driver manually (`sudo pacman -S indi-gpsd`).

waldiniBC, 2024-06-06:
Thanks. I tried to install manually but got no target found from pacman. I am connected to the internet. Not sure if I need to add a repository somewhere to make it work.

[image]

sc74, 2024-06-06:
Are you talking about the indi control panel in Ekos? 

For your GPS to be used, which driver did you select in the hardware manager under others?

MattBlack85, 2024-06-06:
@waldiniBC the GPSD driver should be there

[image]

can you tell me if you see it? I see the panel in indi even if I have gpsd not running

@lhanson not sure how you were able to install indi-gpsd from pacman, there is an AUR package for that but it is very old and it should not be needed, also since it's been compile against an older version of libindi it may be not functional at all

waldiniBC, 2024-06-07:
Yes, there it is right in front of my face. I thought I had exhaustively searched the devices tab. I missed the "Others" item. 
Everything appears to be working correctly now with UBX-M8130 usb gps. Thank you.

MattBlack85, 2024-06-07:
you are welcome @waldiniBC :handshake:

closed 2024-06-07 by MattBlack85