devDucks/astroarch #84: kstars  3.7.1-2 crashed on RPI 4 8GB after AstroArch Update and Dicover Too

Eddy1972N, 2024-06-11:
Help kstars crashed  at te last update pls fix it up 

"Eddy1972N"

MattBlack85, 2024-06-11:
hi @Eddy1972N try to update now, gsl got an update and I had to recompile libindi, kstars and stellarsolver

PS: are you on aarch64 or x64 OS?

Eddy1972N, 2024-06-11:
Hey wow nice Job its run! 
[image]

Eddy1972N, 2024-06-11:
Whats with Discover ??

Eddy1972N, 2024-06-11:
When is the Astro Arch 1.9 rollout ??

MattBlack85, 2024-06-11:
few hours from now - I am just done with the kstars blocker and I am shrinking now the .img

there won't be any need to flash a new image, there are no breaking changes here - if you want to fix disscover run `sudo pacman -S flatpak`

it seems archlinuxarm decided to avoid the usage of discover (or discover doesn't know how to deal with ArchLinuxARM) but flatpak may override this. Although I am not sure how nice flatpak will play with the remaining packages as I never used it on any of my machines

MattBlack85, 2024-06-12:
feel free to re-open should you have any query, I consider this issue closed!