devDucks/astroarch #85: Add SIRIL into the installation image

Olli781, 2024-06-11:
Hello everyone,
I have another comment: does it make sense to include the SIRIL software (https://siril.org/) in the installation package?
Best regards, Oliver

MattBlack85, 2024-06-11:
hey @Olli781, siril is available to be installed already via pacman.

I have no opinion if that should be part of the base image or just to be installed by people that want to use it, but I suppose if few people will react with a :+1: to this issue, we may move it to the base image

Ryther, 2024-06-11:
My 2 cents: I've always seen the project as a mean to have a working kstars/phd2 setup, or at least a setup which main point is to drive the equipment during a session, so SIRIL is not a priority since it's part of the post processing. If you want to use SIRIL to live stack, is the RPI powerful enough to manage both kstars shooting and SIRIL stacking? Also, since SIRIL is available on pacman, it's installation is trivial at best.

MattBlack85, 2025-07-10:
gonna add it in 2.1

MattBlack85, 2025-11-07:
Thanks @BenoitBotton , I think this may be indeed close as the onboarding asks for it now

imahgin, 2025-11-08:
> My 2 cents: I've always seen the project as a mean to have a working kstars/phd2 setup, or at least a setup which main point is to drive the equipment during a session, so SIRIL is not a priority [...]

FYI: Live stacking is recently available in Ekos: https://knro.blogspot.com/2025/08/kstars-v378-is-released.html