devDucks/astroarch #87: AstroMDx update to 2.8.1 and cause the camera

JoyHuang524, 2024-06-12:
Hi,

I tried to update AstroMDx to 2.8.1 as application lunch-up. The updated method is to download the AstroMDx 2.8.1 manual install package from official website and install it. After the AstroMDx updated, I can't find my camera anymore in AstroMDx. I tried to roll back with "**astro-rollback-full**" but seems the AstroMDx still keep as latest version. Can I have the scripting to roll-back the AstroMDx to out of box version? Or when can AstroMDx 2.8.1 being supported in AstroArch?

[image]

Thanks.

B.R./ Joy.

MattBlack85, 2024-06-12:
hi @JoyHuang524 

I don't think the manual update may work as there are issue with dev rules and likely path to things, but devrules first of all as they cannot work with ArchLinux (and are the reason why the one shipped by astrodmx are forcibly lowered in priority to have the ones shipped by indi to work.

I am not even sure if you can cleanly uninstall the old astrodmx now but I would suggest the following things in order:
- `sudo pacman -R astro_dmx`
- sudo find / -name "*dmx*"
- delete by hand all the folders (and files) returned by the previous command
- reinstall astrodms `sudo pacman -S astro_dmx`
- get back to latets Kstars with `update-astroarch` (rollback helpers are only for indi/kstars)

The latest version should come out soon, between today and tomorrow

JoyHuang524, 2024-06-12:
Hi @MattBlack85,

Thanks a lot for your comments. I tried to remove the AstroDMx and re-installed, now is OK.
one information for reference, I have to remove all files in **/opt/AstroDMx-Capture/** to re-install AstroDMx well. 
Thanks again.

B.R./ Joy.

MattBlack85, 2024-06-12:
@JoyHuang524 if you sync now with `sudo pacman -Sy` or `update-astroarch` it should pull version 2.8.1 which has been released few minutes ago

MattBlack85, 2024-06-12:
should be ok, closing this. Feel free to re-open should you have still issues

closed 2024-06-12 by MattBlack85