devDucks/astroarch #90: Gpio pwm boot in autostart how can i used?!

Eddy1972N, 2024-06-16:
I want to use the pwm cooler, but i don't find an order how?! To set on the boot config ! Need help !

MattBlack85, 2024-06-17:
hey @Eddy1972N 

can you send me some links that show how this is done on raspbian or other OS for raspberry? I get you want to use a cooler driven with pulses but I am not sure where to start looking at here

Eddy1972N, 2024-06-17:
[image]

My pwm Cooler : https://amzn.eu/d/imfq2Hj

Thats all ! 

https://tutorials-raspberrypi.de/raspberry-pi-luefter-anschliessen-steuern/?utm_content=cmp-true

https://www.the-diy-life.com/connecting-a-pwm-fan-to-a-raspberry-pi/

Look and try 

How to get on Astro arch linux

MattBlack85, 2024-06-17:
you may try:
- install rpi-gpio `paru -S python-rpi-gpio`

then following the article here
```
git clone https://github.com/mklements/PWMFanControl.git
$ cd PWMFanControl
$ cp FanProportional.py ~/FanProportional.py
$ cp FanStepped.py ~/FanStepped.py
```

and the crontab, just be aware that the user is not `pi` like in the article but `astronaut` so the path to the script must be modified accordingly

Eddy1972N, 2024-06-18:
Sorry it's don't Go the First line

MattBlack85, 2024-06-19:
@Eddy1972N what do you mean "it doesn't go"?
You need to be more specific, what did you run, what did you see

Eddy1972N, 2024-06-20:
Had astro arch not wlan 5ghz Support?? i try 5ghz WLAN ?! But Not conect

closed 2025-09-15 by MattBlack85