devDucks/astroarch #92: Failed to connect to server - raspberry pi zero 2 W

KiwiFPVHarry, 2024-07-19:
Hi,

first of all thank you for the  work..this has been the first view of my Uranus-C being able to connect to teh Pi. Unfortunately all goes well until I try to connect to the server. VNC on screen displays teh Connect button but when clicked it says Failed to connect to server. Hopefully this doesn't mean that it wont be able to run...have tried Astroberry etc, but they all cant either run or have old INDI drivers

monitor connected to Pi shows this..
[image]

Thanks

P.S> this is after I disconnected the Camera

MattBlack85, 2024-07-19:
hi @KiwiFPVHarry and thanks for the report, can you provide the following info?

- astroarch version used
- have you updated recently?
- are you comfortable using ssh for some debugging?

KiwiFPVHarry, 2024-07-19:
Hi Mattia, it is 1.9.0 from your Google drive. I tried to update, but since
the Pi hasn't got a direct way of connecting to the network (WiFi only) and
I didn't know how or where to make the changes to enable elan the update
didn't work. Other than that yes I am happy to use ssh - that's how I tried
to update...thank you
Harry from Aotearoa

MattBlack85, 2024-07-19:
thanks @KiwiFPVHarry for starting I would try to check why x0vncserver failed with `sudo systemctl status x0vncserver`

I may start to think that due to resource constraint (512MB of RAM) there is no room to run the DE

KiwiFPVHarry, 2024-07-20:
Hi Mattia,

I presumed that the RAM might be the limiting factor too..just not sure..anyway here are the outputs from two different commands ..Thanks for looking into it Grazie mille
[image]
[image]

MattBlack85, 2024-07-20:
thanks! At first glance it doesn't look like any RAM issue.
Is X active at all?

`pgrep -a Xorg`

KiwiFPVHarry, 2024-07-20:
Hi Mattia, pgrep -a Xorg had no output at all. Neither did pgreg -l Xorg. I also tried to connect via vnc and there was no connection

MattBlack85, 2024-08-22:
@KiwiFPVHarry I am convinced the hardware constraints are the problem here and the DE has not enough resources to start

KiwiFPVHarry, 2024-08-22:
Thank you for replying..yeah I have come to the same conclusion...will
concentrate more on getting the right OAG and guide camera for now and look
at the IT side after that. Thanks

MattBlack85, 2024-08-22:
any time! closing this issue then