devDucks/astroarch #93: Keine Verbindung zur FRITZ!Box

Eddy1972N, 2024-07-28:
Seit dem letzten Update kann ich keine w-lan Verbindung zur Fritz Box herstellen! Mit der vorherigen Version funktioniert das sehr gut was ist bei euch passiert?? 

Bitte bug Fixing

Eddy1972N, 2024-07-29:
No conect on wlan0 bevor the Update nativ v1.9 is conect the wlan 0 on raspberry pi4 !!! Please kill the bug ! I'm Not amused about that ! Please bugfix!!

MattBlack85, 2024-07-29:
@Eddy1972N I may have a look but the screenshot you sent is a bit unclear. It seems you made a lot of netowors and you seem to run a WiFi adapter in addition to the raspberry WiFi, is this correct?

Eddy1972N, 2024-07-29:
Des i made an Access Point 3of them! But i Cant the Main w-lan From RPI 4 (wlan0 ) Not conect on the router at Home ! On the nativ 1.9 was it Connect ! 

Pleasanton try it make 3 Access Point on the Software! On the 1.9 and than past of the Update From the Server! Any Protokoll is broken ! I think so!

MattBlack85, 2024-07-29:
@Eddy1972N not sure I understand you, are you saying it was working and then you updated the system and it doesn't work anymore?

Eddy1972N, 2024-07-29:
> @Eddy1972N not sure I understand you, are you saying it was working and then you updated the system and it doesn't work anymore?

That right!

Eddy1972N, 2024-07-29:
Thats all i put my Passwort in it but nothing! Its Kick me out

[image]

mistertroy, 2024-07-29:
I've had the same problem. I did an update prior to imaging on Friday night, lost wifi, could only connect to the hotspot, so did it that way friday night. Sat down and tried to sort it out on saturday but couldn't, so reflashed the card. It was able to connect to wifi, so i did another update - then it couldn't. Was having issues with Kstars crashing when ending Ekos anyway, so probably needed a reflash anyway! Am able to look into it again today, so am currently reflashing the SD card.

I'm using a Pi4, running AA1.9. No problems before, kept it mostly bleeding edge. won't be doing that anymore!

mistertroy, 2024-07-30:
And after reflashing it all works. I'm not going to update though until it gets cloudy again and I've backed everything up!

MattBlack85, 2024-07-30:
I will look for any NetowrkManager updates lately, I am not sure at the moment if was any of its updates or simply a kernel that broke it

Eddy1972N, 2024-08-02:
And where is the Bug ???

MattBlack85, 2024-08-02:
I have no idea, that is not something I wrote,  use or mantain. That is a package released from KDE and distributed by the ArchLinuxARM team

Eddy1972N, 2024-08-02:
> I will look for any NetowrkManager updates lately, I am not sure at the moment if was any of its updates or simply a kernel that broke it

Ist nothing change With the last Update at 02.08.24 !!!!

Please Bug fix

sc74, 2024-08-07:
Hi. It seems that the pis are affected by a problem for the brcmfmac driver. View your logs with dmesg -w. If you have something like this

brcmfmac: brcmf_set_channel: set chanspec 0xd0...fail, reason -52

So, try to apply the solution found for this issue

https://github.com/home-assistant/operating-system/issues/3367#issuecomment-2269028464

create a file /etc/modprobe.d/brcmfmac.conf containing this line:

options brcmfmac feature_disable=0x82000

and reboot.

Another thing, if you are using openssh please read the archlinux page 

https://archlinux.org/news/the-sshd-service-needs-to-be-restarted-after-upgrading-to-openssh-98p1/

MattBlack85, 2024-08-22:
closed as fixed in 1.9.2

closed 2024-08-22 by MattBlack85