devDucks/astroarch #94: Wifi problem After update on Raspberry Pi 5

rojofree, 2024-08-06:
Hi, friend. Install astroarch on a raspberry pi5. Everything worked ok. After updating with update-astroarch. It constantly asks me for the WiFi password and I lost connection. I can only connect via cable and I can't find a solution. Any ideas? Thank you! Greetings from Argentina

sc74, 2024-08-07:
Hi. It seems that the pis are affected by a problem for the brcmfmac driver. View your logs with dmesg -w. If you have something like this

brcmfmac: brcmf_set_channel: set chanspec 0xd0...fail, reason -52

So, try to apply the solution found for this issue

https://github.com/home-assistant/operating-system/issues/3367#issuecomment-2269028464

create a file /etc/modprobe.d/brcmfmac.conf containing this line:

options brcmfmac feature_disable=0x82000

and reboot.

Another thing, if you are using openssh please read the archlinux page 

https://archlinux.org/news/the-sshd-service-needs-to-be-restarted-after-upgrading-to-openssh-98p1/

BenoitBotton, 2024-08-10:
I had the exact same issue. thanks @sc74 , your brcmfmac solution worked

rojofree, 2024-08-12:
Hi Friend.  worked. Thanks a lot.

Another question. I bought the argon neo 5 m.2 nvme case that with the latest Raspberry Pi OS I have 800 MB/sec and with astroarch I have 370 MB/sec
What can I do to install? The argon script does not work. It says it only works with Raspberry Pi OS.
This is the case
https://argon40.com/products/argon-neo-5-m-2-nvme-for-raspberry-pi-5

This is the manual.
https://cdn.shopify.com/s/files/1/0556/1660/2177/files/FOR_WEB_INSTRUCTION_MANUAL_NEO_5_M.2_NVMe_20240703.pdf?v=1719989485
This is the script that made it work for me on Raspberry PI OS.
curl https://download.argon40.com/argon-eeprom.sh | bash curl https://download.argon40.com/argonneo5.sh | bash Thanks!

MattBlack85, 2024-08-22:
fixed in 1.9.2 - updating with `update-astroarch` will fix it if the file is not there

closed 2024-08-22 by MattBlack85

JosepPuyol, 2024-08-22:
> fixed in 1.9.2 - updating with `update-astroarch` will fix it if the file is not there

Dear, in my case there is no difference. No wifi. I also tried a fresh install and update it. Same error in dmsg.

MattBlack85, 2024-08-22:
@JosepPuyol have you reboot the system after the update?
Can you check you have the proper file with proper content under /etc/modeprobe.d/brcmfmac.conf

Additionally, how did you update?

JosepPuyol, 2024-08-22:
> @JosepPuyol have you reboot the system after the update?
> Can you check you have the proper file with proper content under /etc/modeprobe.d/brcmfmac.conf
> 
> Additionally, how did you update? 

Yes. First I add manually the file with no results. The final test was flash again astroarch, update, reboot. No wifi, only cable. Rpi5.

JosepPuyol, 2024-08-22:
The file is there. Update with update-astroarch.

MattBlack85, 2024-08-23:
@JosepPuyol I think your issue is not related to this one, can you open a new one posting also kindly:

- dmesg output
- a short video of the issue you are facing

thanks!