devDucks/astroarch #95: 1.9 update and libcamera problems

ripkrizbi, 2024-08-11:
Hi, I've updated to the 1.9 and got reports thet the indi camera driver is failing. I tried diagnosing the libcamera, ran the libcamera-hello, and I got the error about the missing libavcodec.so.60. 
Apparently the avcodec lib was bumped up to 6.1 in the archlinux update and now the libcamera-rpi breaks (and consequently the indi-pylibcamera, which depends on the libcamera-rpi). 
Any help is much appreciated in getting the camera to work again! Many thanks for your continuous support!

MattBlack85, 2024-08-11:
thanks for the report! Having some rest at the moment after a small surgery, if I will find some time, I will recompile the driver in the upcoming week, thanks for the patience in advance!

Mattia

ripkrizbi, 2024-08-11:
Best wishes for the recovery! Take your time!

I investigated a bit more and there seems to be another issue, the indi-pylibcamera depends on libcamera-rpi, which is identical (at least in version) to the installed libcamera, but it triggers the uninstall of libcamera and install of libcamera-rpi, and in the process an outdated python-libcamera is installed which breaks the indi driver. On top of that, the rpicam-apps does not seem to be installed by default with the stack, together with all dependencies. Just to add a bit to the problem :)

Grazie e buon recupero!

MattBlack85, 2024-08-21:
hi @ripkrizbi, I recompiled everything and it seems to be working, whenever you have time to update it woudl be great to get a feedback, thanks!

ripkrizbi, 2024-08-22:
Thank you Mattia!

I did a quick test with a completely fresh install, followed by `update-astroarch`, then `pacman -S indi-pylibcamera`

This prompted me to remove libcamera-* in favour of libcamera-rpi-* and initially it misses to install the `rpicam-apps` and its dependency (you may wish to add it explicitely to PKGBUILD) `libpisp`, which I installed manually.

After that, a quick test with libcamera-hello seems to be working. 

The INDI part does not work (`indiserver -v indi_pylibcamera`), as the python-* packages are all compiled for the version 3.11 and the current python version in the new arch is 3.12, so all the installed dependencies are not visible to indiserver's python.

While you were in recovery I tried to recompile the whole stack myself (Arch is not my home territory :) ) and have made it to work, except some of the python depndencies, which I had to brute-force install via pip (could not get the python stuff to compile :( via PKGBUILD :( ). I have saved this install on a separate SD card and I can look up all the dependencies if you want me to, I think I needed to install additional python packages on top of the ones you provide in the stack. If I remember correctly, the very new release of python-libcamera2 has additional stuff it requires.

Additional question - why do you maintain the separate libcamera-rpi set of packages? The ones that come with the libcamera available through extra repo are more up to date (and you can remove one level of complexity by just using the distro packages if I'm correct)...

Thank you for the great support in resolving this, let me know if I can be of any further help!

MattBlack85, 2024-08-23:
thanks @ripkrizbi you are absolutely right, I missed some deps in PKGBUILD, I packaged it only once (this week was the second) and the procedure is not perfect, thanks to your inputs it will be :)

I will recompile pylibcamera today against py3.12

to your main question, libcamera (https://git.libcamera.org/libcamera/libcamera.git/) officially supports v4l2 cameras but doesn't support specific rpi cameras, that's why the raspi folks mantain a fork (https://github.com/raspberrypi/libcamera) that adds a bunch of raspi camera specifics.

There was an idea (still on the table) to replace completely (and for good) the normal version of libcamera with the one for the raspi. In my mind it makes sense but I am unsure if this won't break other libcamera "normal" stuff and that's why it's not been done so far

ripkrizbi, 2024-08-23:
Many many thanks!

All clear about the libcamera-rpi.
 
Please make sure you recompile all the python-* dependencies for the stack, as I mentioned there were quite a few, and some I was unable to compile as packages :(

Thank you once again for the great support, this really makes a difference!

MattBlack85, 2024-08-23:
@ripkrizbi installing now indi-pylibcamera should bring in anything needed.

I checked the python dependencies but as far as I can tell, the latest versions of the ones I have within my repo have all been compiled against 3.12

ripkrizbi, 2024-08-24:
Thanks Mattia!

unfortunately, some of the deps you maintain are still compiled for 3.11: `simplejpeg`, `av`, `exifread`, 'kmsxx'

I compiled the simplejpeg and bumped it to the most recent version (1.7.4), attached is the PKGBUILD 
[PKGBUILD.simplejpeg.txt](https://github.com/user-attachments/files/16736806/PKGBUILD.simplejpeg.txt)
(I had to compile it within the venv because there is no arch package for python-cmake, I'm sure there is a simple solution to this but I was unable to find it :) )

I'm struggling to compile the av at the moment as there seem to be some incompatibilities with the new ffmpeg. So I had to brute-force it (`pip install av --break-system-packages`). Same for kmsxx  (`pip install rpi-kms --break-system-packages`)

after these updates, I'm able to start the indiserver with the indi_pylibcamera module.

albmef, 2024-09-05:
Hi !
any update ? yesterday i freshly installed EndeavousOS on a old laptop, added your repo and installed the ind-3rdparty-drivers.
i found no indi_libcamera driver for managing a remotely connected HQ pi camera from my PC. i do i fix that ?
thanks

albmef, 2024-09-05:
i tried installing indi-pylibcamera, but it does not work
any idea ? 
the README.MD files says "sudo pacman -S indi-pylibcamera", which results in conflicts whit the local libcamera package...
but how do we make it work ? could you at least give a useful link ?

i tested the HQ camera + PI4_4GB with an ASTROBERRY sdcard and it works with the rpicam driver

here's is what i get when i try to launch the server by hand.
i tried to fix that by installing the missing python packages one by one but i stumble on the missing "simplejpeg" one that i'm not able to fix.

PLEASE HELP

~ » indiserver indi_pylibcamera                                                                                                       astronaut@astroarch
2024-09-05T19:39:06: startup: indiserver indi_pylibcamera
2024-09-05T19:39:06: Driver indi_pylibcamera: Traceback (most recent call last):
2024-09-05T19:39:06: Driver indi_pylibcamera:   File "/usr/bin/indi_pylibcamera", line 33, in <module>
2024-09-05T19:39:06: Driver indi_pylibcamera:     sys.exit(load_entry_point('indi-pylibcamera==2.6.4', 'console_scripts', 'indi_pylibcamera')())
2024-09-05T19:39:06: Driver indi_pylibcamera:              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2024-09-05T19:39:06: Driver indi_pylibcamera:   File "/usr/bin/indi_pylibcamera", line 25, in importlib_load_entry_point
2024-09-05T19:39:06: Driver indi_pylibcamera:     return next(matches).load()
2024-09-05T19:39:06: Driver indi_pylibcamera:            ^^^^^^^^^^^^^^^^^^^^
2024-09-05T19:39:06: Driver indi_pylibcamera:   File "/usr/lib/python3.12/importlib/metadata/__init__.py", line 205, in load
2024-09-05T19:39:06: Driver indi_pylibcamera:     module = import_module(match.group('module'))
2024-09-05T19:39:06: Driver indi_pylibcamera:              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2024-09-05T19:39:06: Driver indi_pylibcamera:   File "/usr/lib/python3.12/importlib/__init__.py", line 90, in import_module
2024-09-05T19:39:06: Driver indi_pylibcamera:     return _bootstrap._gcd_import(name[level:], package, level)
2024-09-05T19:39:06: Driver indi_pylibcamera:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2024-09-05T19:39:06: Driver indi_pylibcamera:   File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
2024-09-05T19:39:06: Driver indi_pylibcamera:   File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
2024-09-05T19:39:06: Driver indi_pylibcamera:   File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
2024-09-05T19:39:06: Driver indi_pylibcamera:   File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
2024-09-05T19:39:06: Driver indi_pylibcamera:   File "<frozen importlib._bootstrap_external>", line 995, in exec_module
2024-09-05T19:39:06: Driver indi_pylibcamera:   File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
2024-09-05T19:39:06: Driver indi_pylibcamera:   File "/usr/lib/python3.12/site-packages/indi_pylibcamera/indi_pylibcamera.py", line 12, in <module>
2024-09-05T19:39:06: Driver indi_pylibcamera:     from picamera2 import Picamera2
2024-09-05T19:39:06: Driver indi_pylibcamera:   File "/usr/lib/python3.12/site-packages/picamera2/__init__.py", line 9, in <module>
2024-09-05T19:39:06: Driver indi_pylibcamera:     from .picamera2 import Picamera2, Preview
2024-09-05T19:39:06: Driver indi_pylibcamera:   File "/usr/lib/python3.12/site-packages/picamera2/picamera2.py", line 26, in <module>
2024-09-05T19:39:06: Driver indi_pylibcamera:     from picamera2.encoders import Encoder, H264Encoder, MJPEGEncoder, Quality
2024-09-05T19:39:06: Driver indi_pylibcamera:   File "/usr/lib/python3.12/site-packages/picamera2/encoders/__init__.py", line 6, in <module>
2024-09-05T19:39:06: Driver indi_pylibcamera:     from .jpeg_encoder import JpegEncoder
2024-09-05T19:39:06: Driver indi_pylibcamera:   File "/usr/lib/python3.12/site-packages/picamera2/encoders/jpeg_encoder.py", line 3, in <module>
2024-09-05T19:39:06: Driver indi_pylibcamera:     import simplejpeg
2024-09-05T19:39:06: Driver indi_pylibcamera: ModuleNotFoundError: No module named 'simplejpeg'
2024-09-05T19:39:06: Driver indi_pylibcamera: read: Connection reset by peer

albmef, 2024-09-07:
Hi 
today i restarted with a fresh 1.9 && updated astroarch
installed libcamera-rpi and libcamera-tools-rpi 
i tried to launch "cam" but there are missing libs.
here are the missing libs i tried to fix
libpisp -> pacman -S libpisp 
libboost-log -> pacman -S boost-libs 

then i'm able to launch cam but there are errors:
~ » cam -l                                                                                                                                                                          astronaut@astroarch
[0:01:22.451106369] [1641] ERROR IPAModule ipa_module.cpp:172 Symbol ipaModuleInfo not found
[0:01:22.451575680] [1641] ERROR IPAModule ipa_module.cpp:292 v4l2-compat.so: IPA module has no valid info
[0:01:22.451711753] [1641]  INFO Camera camera_manager.cpp:313 libcamera v0.3.0
[0:01:22.474326203] [1642] ERROR DmaBufAllocator dma_buf_allocator.cpp:116 Could not open any dma-buf provider
[0:01:22.488454454] [1642]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:01:22.490073739] [1642] ERROR RPI vc4.cpp:215 Failed to register camera imx477 10-001a: -12
Available cameras:

my camera is not seen by cam/qcam, even if i put dtoverlay=imx477 in /boot/config.txt and the cam is detected by the kernel

did i something wrong ?

regards,

MattBlack85, 2024-09-08:
@albmef as @ripkrizbi made me noticing, there are some deps that are broken. Will try to fix them asap to have the package ready to be working with no additional steps

albmef, 2024-09-08:
thanks !
Feel free to ask for any quick/extensive test if needed
i own a PI4_4GB fitted with an HQ cam i can't wait to take nice night sky pictures with ;)
regards,
bob

MattBlack85, 2024-09-09:
@ripkrizbi I got all but exif recompiled, are you able to tell me what package requires it and which version? I think we are using exifread but just noticed there is also an exif package and got confused

ripkrizbi, 2024-09-09:
@MattBlack85 first of all, thank you for all the effort in maintaining the distro and the extra packages! I'm fully aware how much effort it takes to do such a thing, especially on a voluntary basis, and just want to tell you how much I appreciate it! 

I quickly checked, and can confirm that the packages I needed to make the indi_pylibcamera driver work were the `simplejpeg`, `av `and `kmsxx`. As I was going through the packages you maintain I noticed that the `exifread` was still in 3.11 so I included it into the list, thinking that it may be a dependency for something else. 

So, at least for my issue, `exifread` is not required.

On a side note, how frequent are your bleeding-edge libindi/drivers/kstars builds? Do you have an automated process for periodic rebuilds? I'm trying to do a fresh git-based recompile because there may be a fix for my mount in the indilib after the official 2.0.9...

albmef, 2024-09-09:
Hi 
thanks for your kind support
today i upgraded arch and the new boost lib version 1.86.0 broke rpicam-apps
(see my new issue)
currently i'm performing some test with HQ camera, for my astro setup, so i needed a quick and dirty fix
regards,
bob

MattBlack85, 2024-09-09:
@ripkrizbi thanks for your words 🙇

For bleeding edge, I don't push it very often, I don't have enough horse power with my building machine to have it running periodically. My build box is a pi4.
The best thing would be to have a github actions running daily and in theory, that could work but only for indi. Kstars is a beast and it takes hours to compile on github actions or a raspberrypi 4.
I workaround that when I package the new official versions by using distcc but that's it, no fancy automation😋

I am planning to replace my build box with 2/3 Raspi 5 2Gi, that would probably help since the cluster would run 24/7 and have 16 powerful arm cpu to use, but when it is gonna happen I really have no idea 🫠

albmef, 2024-09-20:
hi @ripkrizbi 
could you provide me a quick Howto guide for making the indi_pylibcamera to run ? i'm really looking forward this.
this morning i tried to compile simplejpeg but no luck. i'm lacking the knowledge and time for fixing that.
the pi HQ cam is part of my setup i plan to use it as guiding cam, so a quick and dirty hack is ok for me in the meantime.
regards,
bob

albmef, 2024-09-20:
sorry but it does not work yet
2024-09-20T09:24:42: startup: indiserver indi_pylibcamera
2024-09-20T09:24:43: Driver indi_pylibcamera: Traceback (most recent call last):
2024-09-20T09:24:43: Driver indi_pylibcamera:   File "/usr/lib/python3.12/site-packages/picamera2/previews/drm_preview.py", line 9, in <module>
2024-09-20T09:24:43: Driver indi_pylibcamera:     import kms as pykms
2024-09-20T09:24:43: Driver indi_pylibcamera: ModuleNotFoundError: No module named 'kms'

MattBlack85, 2024-09-20:
have you updated recently @albmef ?

albmef, 2024-09-20:
yes system up to date. i do it everyday
i installed the dependencies by hand to be sure
sudo pacman -S python-av kmsxx-git python-pillow python-pycamera2 python-lxml python-astropy python-pidng python-v4l2 python-simplejpeg libcamera-rpi rpicam-apps

but no luck

MattBlack85, 2024-09-20:
Interesting thanks!, We got the camera working flawlessly on @FireNWater somehow, will have a look at what we did additionally

albmef, 2024-09-20:
thanks 
but there is no python-kmsxx package in the arch repos
bob

ripkrizbi, 2024-09-20:
@albmef as I have clearly written, the said package is maintained by @MattBlack85 and it is located in the `astromatto` repo. Have you tried running the commands I gave?

albmef, 2024-09-20:
i tried force pip install rpi-kms but this does not work either
2024-09-10T23:18:22: startup: indiserver indi_pylibcamera
2024-09-10T23:18:26: Driver indi_pylibcamera: Traceback (most recent call last):
2024-09-10T23:18:26: Driver indi_pylibcamera:   File "/usr/lib/python3.12/site-packages/picamera2/previews/drm_preview.py", line 9, in <module>
2024-09-10T23:18:26: Driver indi_pylibcamera:     import kms as pykms
2024-09-10T23:18:26: Driver indi_pylibcamera: ModuleNotFoundError: No module named 'kms'

albmef, 2024-09-20:
hope this helps
ldd /usr/lib/python3.12/site-packages/pykms/pykms.so                                                                                                                                astronaut@astroarch
	linux-vdso.so.1 (0x0000ffff83687000)
	libkms++.so.0 => /usr/lib/libkms++.so.0 (0x0000ffff83520000)
	libkms++util.so.0 => /usr/lib/libkms++util.so.0 (0x0000ffff834f0000)
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x0000ffff83240000)
	libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x0000ffff83200000)
	libc.so.6 => /usr/lib/libc.so.6 (0x0000ffff83040000)
	/usr/lib/ld-linux-aarch64.so.1 (0x0000ffff8364a000)
	libdrm.so.2 => /usr/lib/libdrm.so.2 (0x0000ffff83000000)
	libfmt.so.10 => not found
	libm.so.6 => /usr/lib/libm.so.6 (0x0000ffff82f50000)

albmef, 2024-09-20:
the problem is that i doubt that "pip install rpi-kms --break-system-packages" actually worked
ls /usr/lib/python3.12/site-packages/rpi*                                                                                                                                           
zsh: no matches found: /usr/lib/python3.12/site-packages/rpi*

how do i check if the files were installed properly ?

ripkrizbi, 2024-09-20:
If you installed without sudo, then the library should be in your `~/.local/lib/python3.12/site-packages` directory, check there.

also, start python, then
`import pykms`

and if you get no errors then it's there. Make sure you start indiserver with the same user privileges as the ones you used to do the pip install

albmef, 2024-09-20:
pip with sudo

sudo pip install rpi-kms --break-system-packages                                                                                                                                    
[sudo] password for astronaut: 
DEPRECATION: Loading egg at /usr/lib/python3.12/site-packages/urjtag-2021.3-py3.12-linux-aarch64.egg is deprecated. pip 24.3 will enforce this behaviour change. A possible replacement is to use pip for package installation. Discussion can be found at https://github.com/pypa/pip/issues/12330
Collecting rpi-kms
  Downloading rpi_kms-0.1a1.tar.gz (3.4 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: rpi-kms
  Building wheel for rpi-kms (pyproject.toml) ... done
  Created wheel for rpi-kms: filename=rpi_kms-0.1a1-cp312-cp312-linux_aarch64.whl size=227855 sha256=f69514854ae9775b487384492ceb2fac1fa70caa3a54402da17e2c89048a8c0b
  Stored in directory: /root/.cache/pip/wheels/c7/a9/5b/1c7a9b1f26df2f341d80cdc0bef9a19bf7236f110565832d4c
Successfully built rpi-kms
Installing collected packages: rpi-kms
Successfully installed rpi-kms-0.1a1
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager, possibly rendering your system unusable.It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv. Use the --root-user-action option if you know what you are doing and want to suppress this warning.
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

check if installed

~ » tree /usr/lib/python3.12/site-packages/ | grep kms                                                                                                                                  
├── pykms
│   ├── pykms.cpython-312-aarch64-linux-gnu.so
│   └── pykms.so
├── rpi_kms-0.1a1.dist-info
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

test with python

~ » python                                                                                                                                                                              
Python 3.12.6 (main, Sep 13 2024, 13:27:21) [GCC 14.1.1 20240507] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import kms
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'kms'
>>> import pykms
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/astronaut/.local/lib/python3.12/site-packages/pykms/__init__.py", line 1, in <module>
    from .pykms import *
ImportError: libfmt.so.10: cannot open shared object file: No such file or directory
>>> 

test indiserver with sudo

 » sudo indiserver indi_pylibcamera                                                                                                                                                1 
2024-09-20T11:14:38: startup: indiserver indi_pylibcamera
2024-09-20T11:14:39: Driver indi_pylibcamera: Traceback (most recent call last):
2024-09-20T11:14:39: Driver indi_pylibcamera:   File "/usr/lib/python3.12/site-packages/picamera2/previews/drm_preview.py", line 9, in <module>
2024-09-20T11:14:39: Driver indi_pylibcamera:     import kms as pykms
2024-09-20T11:14:39: Driver indi_pylibcamera: ModuleNotFoundError: No module named 'kms'

ripkrizbi, 2024-09-20:
It appears you have installed the rpi-kms package twice, and now you have confused python somehow. In your error when importing pykms within python, you can see that the path is a LOCAL path (ie rpi-kms installed without sudo, `File "**/home/astronaut/.local/lib**/python3.12/site-packages/pykms...`).

remove both packages through pip (both local and system), then install only local (as I have written, please read the suggestions thoroughly). Then try again.

albmef, 2024-09-20:
as you said i removed both local and system using pip uninstall

so only kmsxx-git remains on the system

~ » tree /usr/lib/python3.12/site-packages/ | grep kms                                                                                                                                  astronaut@astroarch
├── pykms
│   └── pykms.so

~ » tree .local/lib/python3.12/site-packages | grep kms                                                                                                                                 
~ »       

i cannot remove the kmsxx-git package because it break the dependency of indi-pylibcamera

~ » sudo pacman -R kmsxx-git                                                                                                                                                        1 
[sudo] password for astronaut: 
checking dependencies...
error: failed to prepare transaction (could not satisfy dependencies)
:: removing kmsxx-git breaks dependency 'kmsxx-git' required by indi-pylibcamera

so i leave it installed.

i test with python

~ » python                                                                                                                                                                              
Python 3.12.6 (main, Sep 13 2024, 13:27:21) [GCC 14.1.1 20240507] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pykms
>>> import kms
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'kms'
>>> 

pykms is found but still no kms module

now i reinstall rpi-kms system wide with SUDO

~ » sudo pip install rpi-kms --break-system-packages                                                                                                                                
DEPRECATION: Loading egg at /usr/lib/python3.12/site-packages/urjtag-2021.3-py3.12-linux-aarch64.egg is deprecated. pip 24.3 will enforce this behaviour change. A possible replacement is to use pip for package installation. Discussion can be found at https://github.com/pypa/pip/issues/12330
Collecting rpi-kms
  Using cached rpi_kms-0.1a1-cp312-cp312-linux_aarch64.whl
Installing collected packages: rpi-kms
Successfully installed rpi-kms-0.1a1
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager, possibly rendering your system unusable.It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv. Use the --root-user-action option if you know what you are doing and want to suppress this warning.

i check it

~ » tree /usr/lib/python3.12/site-packages/ | grep kms                                                                                                                                  
├── pykms
│   ├── pykms.cpython-312-aarch64-linux-gnu.so
│   └── pykms.so
├── rpi_kms-0.1a1.dist-info

~ » python                                                                                                                                                                              
Python 3.12.6 (main, Sep 13 2024, 13:27:21) [GCC 14.1.1 20240507] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pykms
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.12/site-packages/pykms/__init__.py", line 1, in <module>
    from .pykms import *
ImportError: libfmt.so.10: cannot open shared object file: No such file or directory
>>> import kms
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'kms'

it seems that there is still a conflict 
but i'm not able to resolve it because indi-pylibcamera depends on kmsxx-git and i cannot remove it without removing indi-pylibcamera.
sorry but i don't really understand the ins and outs of forcing rpi-kms with pip, and second, which package actually provides the expected kms module (not pykms which is provided by kmsxx-git)
any idea ?

albmef, 2024-09-20:
okay i renamed the pykms module for removing the conflict
~ » sudo mv /usr/lib/python3.12/site-packages/pykms/pykms.so /usr/lib/python3.12/site-packages/pykms/pykms.so.bak                                                                       astronaut@astroarch
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
~ » tree /usr/lib/python3.12/site-packages/ | grep kms                                                                                                                                   
├── pykms
│   ├── pykms.cpython-312-aarch64-linux-gnu.so
│   └── pykms.so.bak
├── rpi_kms-0.1a1.dist-info

now it seems that there is a runtime lib dependency issue with the rpi-kms package

indiserver throws me this error

2024-09-20T12:47:40: Driver indi_pylibcamera:     from .pykms import *
2024-09-20T12:47:40: Driver indi_pylibcamera: ImportError: libfmt.so.10: cannot open shared object file: No such file or directory

~ » ldd /usr/lib/python3.12/site-packages/pykms/pykms.cpython-312-aarch64-linux-gnu.so                                                                                             
	linux-vdso.so.1 (0x0000ffff9a043000)
	libkms++.so.0 => /usr/lib/libkms++.so.0 (0x0000ffff99ef0000)
	libkms++util.so.0 => /usr/lib/libkms++util.so.0 (0x0000ffff99ec0000)
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x0000ffff99c10000)
	libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x0000ffff99bd0000)
	libc.so.6 => /usr/lib/libc.so.6 (0x0000ffff99a10000)
	/usr/lib/ld-linux-aarch64.so.1 (0x0000ffff9a006000)
	libdrm.so.2 => /usr/lib/libdrm.so.2 (0x0000ffff999d0000)
	libfmt.so.10 => not found
	libm.so.6 => /usr/lib/libm.so.6 (0x0000ffff99920000)

please let me know when you have access to real hardware for doing actual tests. in the meantime i will try with a fresh install. please excuse my lack of knowledge with PKGBUILD. i'm an seasoned hardware guy, not software
as i said i can do tests at my side if needed.
thanks for your support
bob

ripkrizbi, 2024-09-21:
@albmef I have the real hardware and was able to set the camera up no problem, so:

**Start from fresh installation**, then run the following commands in the **exact order and without running or installing anything else**:
 - `update-astroarch`
 - `sudo pacman -S indi-pylibcamera`
 - `pip install rpi-kms --break-system-packages` (**without sudo!**)

after that, `indiserver -v indi_pylibcamera` should not complain about anything and the camera should be ready to use. Report back if this is not the case, but please do not install or run anything else other than the above.

albmef, 2024-09-21:
hi 
i did exactly what you said today from a fresh 2.0.9 image and i still get this missing kms error
im' sorry

albmef, 2024-09-21:
did it  !
i rebuilt the package from here
https://gitlab.archlinux.org/archlinux/packaging/packages/fmt/-/tree/10.2.0-1?ref_type=tags

installed the deps (doxygen,...)
i changed to aarch64 in the PKGBUILD file, makepkg, 
then removed doxygen and spdlog (depend both on fmt v11)
then sudo pacman -U fmt-10.2.0-2-aarch64.pkg.tar.xz 
added IgnorePkg = fmt in /etc/pacman.conf

now indiserver starts and i'm able to use the pi camera HQ !
hope this helps, thanks for your kind support and patience.
bob

sc74, 2024-09-22:
The recent update of the doxygen and fmt packages breaks the kmsxx-git and libcamera packages. Packages will be updated soon. Be careful of your modifications which could break your systems. For Arch, it is preferable to use a PKGBUILD that can be uninstalled or updated properly and avoid 'pip install'

albmef, 2024-09-22:
hi @MattBlack85 @ripkrizbi 

i managed to get it work with fmt v11 so far
here's what i did:

- upgrade fmt to v11  (removed exception in pacman.conf then sudo pacman -Su)
- pip uninstall --break-system-packages rpi-kms    (removed broken local pykms)
- sudo pacman -R kmsxx-git indi-pylibcamera     (removed broken system libkms++ libkms++-utils and pykms)
- clone and rebuild kmsxx-git from https://aur.archlinux.org/packages/kmsxx-git   (changed to pkgrel=3)
- sudo pacman -U kmsxx-git-r668.9ae90ce-3-aarch64.pkg.tar.xz   (installed the built package)
- sudo pacman -S indi-pylibcamera  

- test pykms lib deps
ldd /usr/lib/python3.12/site-packages/pykms/pykms.so 
        linux-vdso.so.1 (0x0000ffff85b06000)
        libkms++.so.0 => /usr/lib/libkms++.so.0 (0x0000ffff85990000)
        libkms++util.so.0 => /usr/lib/libkms++util.so.0 (0x0000ffff85950000)
        libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x0000ffff856a0000)
        libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x0000ffff85660000)
        libc.so.6 => /usr/lib/libc.so.6 (0x0000ffff854a0000)
        /usr/lib/ld-linux-aarch64.so.1 (0x0000ffff85ac9000)
        libdrm.so.2 => /usr/lib/libdrm.so.2 (0x0000ffff85460000)
        libfmt.so.11 => /usr/lib/libfmt.so.11 (0x0000ffff85410000)
        libm.so.6 => /usr/lib/libm.so.6 (0x0000ffff85360000)

tested in kstar/EKOS -> ok

hope this helps, regards
bob

MattBlack85, 2024-09-22:
cool, let me rebuild kmsxxx then, should be ready in a few

MattBlack85, 2024-09-22:
this is done now, kmsxxx rel3 is officially available from my repo now

albmef, 2024-09-22:
please be careful @MattBlack85 , as if you performed the build without having done the pacman -R kmsxx-git before, the libkms++-utils lib will not have been built with the right version of fmt (i did the mistake myself), so:

please do the test in your clone dir after the makepkg
- cd pkg/kmsxx-git   
- /pkg/kmsxx-git (master*) » ldd pkg/kmsxx-git/usr/lib/libkms++util.so    
        linux-vdso.so.1 (0x0000ffffba7c7000)
        libkms++.so.0 => /usr/lib/libkms++.so.0 (0x0000ffffba6d0000)
        libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x0000ffffba420000)
        libm.so.6 => /usr/lib/libm.so.6 (0x0000ffffba370000)
        libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x0000ffffba330000)
        libc.so.6 => /usr/lib/libc.so.6 (0x0000ffffba170000)
        /usr/lib/ld-linux-aarch64.so.1 (0x0000ffffba78a000)
        libdrm.so.2 => /usr/lib/libdrm.so.2 (0x0000ffffba130000)
        libfmt.so.11 => /usr/lib/libfmt.so.11 (0x0000ffffba0e0000)

- /pkg/kmsxx-git (master*) » ldd pkg/kmsxx-git/usr/lib/libkms++.so     
        linux-vdso.so.1 (0x0000ffffaf344000)
        libdrm.so.2 => /usr/lib/libdrm.so.2 (0x0000ffffaf250000)
        libfmt.so.11 => /usr/lib/libfmt.so.11 (0x0000ffffaf200000)
        libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x0000ffffaef50000)
        libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x0000ffffaef10000)
        libc.so.6 => /usr/lib/libc.so.6 (0x0000ffffaed50000)
        /usr/lib/ld-linux-aarch64.so.1 (0x0000ffffaf307000)
        libm.so.6 => /usr/lib/libm.so.6 (0x0000ffffaeca0000)

/pkg/kmsxx-git (master*) » ldd pkg/kmsxx-git/usr/lib/python3.12/site-packages/pykms/pykms.so  
        linux-vdso.so.1 (0x0000ffff80c0b000)
        libkms++.so.0 => /usr/lib/libkms++.so.0 (0x0000ffff80a90000)
        libkms++util.so.0 => /usr/lib/libkms++util.so.0 (0x0000ffff80a50000)
        libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x0000ffff807a0000)
        libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x0000ffff80760000)
        libc.so.6 => /usr/lib/libc.so.6 (0x0000ffff805a0000)
        /usr/lib/ld-linux-aarch64.so.1 (0x0000ffff80bce000)
        libdrm.so.2 => /usr/lib/libdrm.so.2 (0x0000ffff80560000)
        libfmt.so.11 => /usr/lib/libfmt.so.11 (0x0000ffff80510000)
        libm.so.6 => /usr/lib/libm.so.6 (0x0000ffff80460000)

bob

MattBlack85, 2024-09-22:
Bob, the rel3 just shipped is dynamically linked to libfmt.so.11

albmef, 2024-09-22:
yes but with the old one still on my system, the built did performed as expected so i had to completely remove any occurence of libkms++ and pykms

MattBlack85, 2024-09-22:
correct, after that bump you should have nothing installed additionally by hand or they may conflict with the actual paths used by our stuff

albmef, 2024-09-22:
tested ok after pacman -Suy and reboot 
capture with EKOS : OK
rpicam-hello : OK
@ripkrizbi that solved my issue with HQ camera and indi

note that i'm now able to set the shutter time from 1ms to 600s, and read sensor temperature
YAY !!!!

astroberry cannot do that with the HQ camera because it's so old... the shutter speed is limited to few seconds or so, and old raspistill app sucks ! XD

thanks all for your support
best regards,
bob

MattBlack85, 2024-09-22:
thanks everyone involved into fixing this issue! It bugged us heavily in this month but I am super happy we solved it as a community, I really appreciate all you did folks.

Thanks @ripkrizbi @albmef @sc74 @FireNWater

albmef, 2024-09-22:
yes it definitely worth having a reliable and FOSS solution for astro

MattBlack85, 2024-09-29:
closing this as solved :)

closed 2024-09-29 by MattBlack85