devDucks/astroarch #99: Wifi problem After update on Raspberry Pi 5 (again)

JosepPuyol, 2024-08-23:
@MattBlack85 thanks for your comments.

The issue #94 is closed, but I have no solution. Sorry for that.

Working perfectly for months, artroarch is awesome! After update to 1.9  with update-astroarch  I can only connect via cable. 

These are the unsuccessful tests I made:

1- Add the file manually, reboot.
2- Delete the file and update to 1.9.2 with update-astroarch, reboot.
3- Flash a new card with astroarch, update, and reboot.

No success in all the cases. It cannot connect to other wifi networks and it is not possible to reach the hotspot from another computer.

Here is my dmesg file 
[dmesg-astroarch.txt](https://github.com/user-attachments/files/16729656/dmesg-astroarch.txt)

Is there is somebody in a similar situation? I appreciate any help you can provide. Thanks in advance.

MattBlack85, 2024-08-23:
thanks @JosepPuyo,l hope I'll be able to find something for your case

MattBlack85, 2024-08-23:
@JosepPuyol can you give back the output of this command?
` lsmod | grep brcmfmac`

JosepPuyol, 2024-08-23:
Of course.

brcmfmac_wcc           12288  0
brcmfmac              364544  1 brcmfmac_wcc
brcmutil               20480  1 brcmfmac
cfg80211             1003520  1 brcmfmac

MattBlack85, 2024-08-23:
it seems the same as mine, they are identical. Are you trying to connect to an access point that has WPA3 or mixed WPA2/WPA3 mode enabled?

JosepPuyol, 2024-08-23:
I think so.

MattBlack85, 2024-08-23:
found this topic https://github.com/raspberrypi/linux/issues/4718, are you able to test against a network that has only WPA2 so we can exclude it's not it?

JosepPuyol, 2024-08-23:
I can try it tomorrow. Thanks Again!

ripkrizbi, 2024-08-23:
There is a [thread here](https://bbs.archlinux.org/viewtopic.php?id=298025) where people fixed it with downgrading wpa_supplicant to 2.10-8, 

You can get the old version here: https://alaa.ad24.cz/packages/w/wpa_supplicant/

JosepPuyol, 2024-08-24:
@MattBlack85 and @ripkrizbi , dears. 

Before limiting my network to WPA2 only, I checked the network configuration of the Rpi5. Surprisingly, when I try to connect to a new network it chooses WPA3 security. I have changed to WPA2, and it is working now.

It has not been necessary to try downgrading wpa_supplicant to 2.10-8.

Thanks for your kindness. Yours.

MattBlack85, 2024-08-24:
awesome, happy it works friend!

MattBlack85, 2024-09-09:
closing this as done