devDucks/astroarch #102: Instructions to boot from NVME SSD drive

Astrostecco, 2024-09-01:
Ciao Mattia,

I tried the instructions referenced in the Github website, but it seems like there is a problem already with the 1st line of code: the system doesn't find the files to download (see attached image).

Any suggestion on how to proceed?

Thanks!
Stefano
[image]

MattBlack85, 2024-09-01:
@Astrostecco that's a long issue with archlinux mirrors, try first to update the package database with `sudo pacman -Sy` and if fails again with 404, try few times, it should go

Astrostecco, 2024-09-01:
it worked, thank you so much!! :)

teoteo, 2024-09-02:
I had the same issue. I fixed editing the file /etc/pacman.d/mirrorlist:
`sudo nano /etc/pacman.d/mirrorlist`
replacing the line:
`Server = http://mirror.archlinuxarm.org/$arch/$repo`
with:
`Server = http://eu.mirror.archlinuxarm.org/$arch/$repo`
getting it from the [mirror list page](https://archlinuxarm.org/about/mirrors)

Should I get back to mirror.archlinuxarm.org in future?

MattBlack85, 2024-09-08:
@teoteo probably not, ther main mirror which is supposed to load balance based on region still redirects to stuff that doesn't work

closed 2024-09-08 by MattBlack85