devDucks/astroarch #104: 'Could not open any dmaHeap device' FIX

albmef, 2024-09-08:
Hi !
using last 1.9 image (up to date) on a PI4_4GB with HQ camera.

the rpicam-apps need to be launch with sudo due to an access permission error to dma_heap device
i propose a new udev rule to be added.

step to reproduce :
~ » rpicam-hello -n -t 1                                                                                                                                                                astronaut@astroarch
Could not open any dmaHeap device
[2:40:33.915193935] [4981] ERROR IPAModule ipa_module.cpp:172 Symbol ipaModuleInfo not found
[2:40:33.915225268] [4981] ERROR IPAModule ipa_module.cpp:292 v4l2-compat.so: IPA module has no valid info
[2:40:33.915280767] [4981]  INFO Camera camera_manager.cpp:313 libcamera v0.3.0
[2:40:33.934882193] [4982] ERROR DmaBufAllocator dma_buf_allocator.cpp:116 Could not open any dma-buf provider
[2:40:33.955290260] [4982]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[2:40:33.957514762] [4982] ERROR RPI vc4.cpp:215 Failed to register camera imx477 10-001a: -12
ERROR: *** no cameras available ***

if we launch with sudo we get:
~ » sudo rpicam-hello -n -t 1                                                                                                                                                     130 ↵ astronaut@astroarch
[2:40:23.408419338] [4968] ERROR IPAModule ipa_module.cpp:172 Symbol ipaModuleInfo not found
[2:40:23.408504393] [4968] ERROR IPAModule ipa_module.cpp:292 v4l2-compat.so: IPA module has no valid info
[2:40:23.408565522] [4968]  INFO Camera camera_manager.cpp:313 libcamera v0.3.0
[2:40:23.447409709] [4969]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[2:40:23.449793950] [4969]  INFO RPI vc4.cpp:446 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media2 and ISP device /dev/media1
[2:40:23.449894283] [4969]  INFO RPI pipeline_base.cpp:1104 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Mode selection for 2028:1520:12:P
    SRGGB10_CSI2P,1332x990/0 - Score: 3456.22
    SRGGB12_CSI2P,2028x1080/0 - Score: 1083.84
    SRGGB12_CSI2P,2028x1520/0 - Score: 0
    SRGGB12_CSI2P,4056x3040/0 - Score: 887
[2:40:23.450712629] [4968]  INFO Camera camera.cpp:1183 configuring streams: (0) 2028x1520-YUV420 (1) 2028x1520-SBGGR12_CSI2P
[2:40:23.451058996] [4969]  INFO RPI vc4.cpp:621 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC

QUICK FIX derived from this thread:
[https://raspberrypi.stackexchange.com/questions/141106/how-to-fix-the-libcamera-error-could-not-open-any-dmaheap-device](url)

add an udev rule in /etc/udev/rules.d/98-picamera.rules
add the following rule:
SUBSYSTEM=="dma_heap", GROUP="video", MODE="0660"

save, close, then reload the rules (no need to reboot)
sudo udevadm control --reload-rules && sudo udevadm trigger

now retry without sudo:
~ » rpicam-hello -n -t 1                                                                                                                                                                astronaut@astroarch
[2:50:47.813489346] [5363] ERROR IPAModule ipa_module.cpp:172 Symbol ipaModuleInfo not found
[2:50:47.813586086] [5363] ERROR IPAModule ipa_module.cpp:292 v4l2-compat.so: IPA module has no valid info
[2:50:47.813667196] [5363]  INFO Camera camera_manager.cpp:313 libcamera v0.3.0
[2:50:47.853982706] [5364]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[2:50:47.856210153] [5364]  INFO RPI vc4.cpp:446 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media2 and ISP device /dev/media1
[2:50:47.856316078] [5364]  INFO RPI pipeline_base.cpp:1104 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Mode selection for 2028:1520:12:P
    SRGGB10_CSI2P,1332x990/0 - Score: 3456.22
    SRGGB12_CSI2P,2028x1080/0 - Score: 1083.84
    SRGGB12_CSI2P,2028x1520/0 - Score: 0
    SRGGB12_CSI2P,4056x3040/0 - Score: 887
[2:50:47.857093164] [5363]  INFO Camera camera.cpp:1183 configuring streams: (0) 2028x1520-YUV420 (1) 2028x1520-SBGGR12_CSI2P
[2:50:47.857437273] [5364]  INFO RPI vc4.cpp:621 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC

TADAAAWWWWWW !

regards,
bob

MattBlack85, 2024-09-09:
I pack some dev rules for the PI already, I will add this as well so it will work out of the box

albmef, 2024-09-09:
thanks 
in which package did you put the new udev rules please ?
bob

closed 2024-09-09 by albmef