devDucks/astroarch #105: rpicam-apps broken dependency with liboost_program_options 1.83.0

albmef, 2024-09-09:
Hi,
i got an issue after upgrading today. rpicam-apps does not work anymore because of a missing boost lib.

steps to reproduce:
-update databases with sudo pacman -Sy

- list upgradables packages with sudo sudo pacman -Qu
~ » sudo pacman -Qu                                                                                                                                                                     
audit 4.0.2-1 -> 4.0.2-2
base-devel 1-1 -> 1-2
bison 3.8.2-6 -> 3.8.2-7
boost 1.83.0-9 -> 1.86.0-2.  <<<<<<<<<<<<<<<
boost-libs 1.83.0-9 -> 1.86.0-2 <<<<<<<<<<<<<<<
cairo 1.18.0-2 -> 1.18.2-1
chromium 128.0.6613.113-1 -> 128.0.6613.119-1
....

 boost version bumped from 1.83.0 to 1.86.0

-upgrade with sudo pacman -Su 

-launch rpicam-hello
 » rpicam-hello                                                                                                                                                                  127 ↵ 
rpicam-hello: error while loading shared libraries: libboost_program_options.so.1.83.0: cannot open shared object file: No such file or directory

-inpect pacman cache:
» ls -l /var/cache/pacman/pkg/boost-*                                                                                                                                                 
-rw-r--r-- 1 root root 14703844 Jun 26 11:44 /var/cache/pacman/pkg/boost-1.83.0-9-aarch64.pkg.tar.xz
-rw-r--r-- 1 root root      566 Jun 26 11:45 /var/cache/pacman/pkg/boost-1.83.0-9-aarch64.pkg.tar.xz.sig
-rw-r--r-- 1 root root 15072496 Sep  7 03:30 /var/cache/pacman/pkg/boost-1.86.0-2-aarch64.pkg.tar.xz
-rw-r--r-- 1 root root      566 Sep  7 03:30 /var/cache/pacman/pkg/boost-1.86.0-2-aarch64.pkg.tar.xz.sig
-rw-r--r-- 1 root root  2243264 Jun 26 11:44 /var/cache/pacman/pkg/boost-libs-1.83.0-9-aarch64.pkg.tar.xz
-rw-r--r-- 1 root root      566 Jun 26 11:45 /var/cache/pacman/pkg/boost-libs-1.83.0-9-aarch64.pkg.tar.xz.sig
-rw-r--r-- 1 root root  2308108 Sep  7 03:30 /var/cache/pacman/pkg/boost-libs-1.86.0-2-aarch64.pkg.tar.xz
-rw-r--r-- 1 root root      566 Sep  7 03:30 /var/cache/pacman/pkg/boost-libs-1.86.0-2-aarch64.pkg.tar.xz.sig

-downgrade boost packages to 1.83.0-9
(https://wiki.archlinux.org/title/Downgrading_packages)
~ » sudo pacman -U /var/cache/pacman/pkg/boost-1.83.0-9-aarch64.pkg.tar.xz /var/cache/pacman/pkg/boost-libs-1.83.0-9-aarch64.pkg.tar.xz                                                 
[sudo] password for astronaut: 
loading packages...
warning: downgrading package boost (1.86.0-2 => 1.83.0-9)
warning: downgrading package boost-libs (1.86.0-2 => 1.83.0-9)
resolving dependencies...
looking for conflicting packages...

Packages (2) boost-1.83.0-9  boost-libs-1.83.0-9

Total Installed Size:  194.83 MiB
Net Upgrade Size:        2.18 MiB

:: Proceed with installation? [Y/n] 

-test rpicam-hello
~ » rpicam-hello -t 1 -n                                                                                                                                                         [0:38:50.473174687] [3866] ERROR IPAModule ipa_module.cpp:172 Symbol ipaModuleInfo not found
[0:38:50.473233039] [3866] ERROR IPAModule ipa_module.cpp:292 v4l2-compat.so: IPA module has no valid info
[0:38:50.473289964] [3866]  INFO Camera camera_manager.cpp:313 libcamera v0.3.0
[0:38:50.515619151] [3868]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:38:50.517872891] [3868]  INFO RPI vc4.cpp:446 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media2 and ISP device /dev/media0
[0:38:50.517965779] [3868]  INFO RPI pipeline_base.cpp:1104 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Mode selection for 2028:1520:12:P
    SRGGB10_CSI2P,1332x990/0 - Score: 3456.22
    SRGGB12_CSI2P,2028x1080/0 - Score: 1083.84
    SRGGB12_CSI2P,2028x1520/0 - Score: 0
    SRGGB12_CSI2P,4056x3040/0 - Score: 887
[0:38:50.518801680] [3866]  INFO Camera camera.cpp:1183 configuring streams: (0) 2028x1520-YUV420 (1) 2028x1520-SBGGR12_CSI2P
[0:38:50.519906763] [3868]  INFO RPI vc4.cpp:621 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC

works OK now

- temporarily ignore upgrade for the faulty packages:
add the following in /etc/pacman.conf
IgnorePkg   = boost
IgnorePkg   = boost-libs

- check if  exception is working
~ » sudo pacman -Sy                                                                                                                                                                     
:: Synchronizing package databases...
 astromatto is up to date
 core is up to date
 extra is up to date
 community is up to date
 alarm is up to date
 aur is up to date
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
~ » pacman -Qu                                                                                                                                                                          astronaut@astroarch
boost 1.83.0-9 -> 1.86.0-2 [ignored]
boost-libs 1.83.0-9 -> 1.86.0-2 [ignored]
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
~ » sudo pacman -Su                                                                                                                                                                     astronaut@astroarch
:: Starting full system upgrade...
warning: boost: ignoring package upgrade (1.83.0-9 => 1.86.0-2)
warning: boost-libs: ignoring package upgrade (1.83.0-9 => 1.86.0-2)
 there is nothing to do

hope this temporary workaround will help
bob

albmef, 2024-09-09:
hi  @MattBlack85 
thanks for your support
i just upraded it seems to be working again.
regards,
bob

closed 2024-09-09 by albmef

OLEGsandrMB, 2025-04-22:
Hi.
I need your help .
With fresh install, followed by `update-astroarch`, then `pacman -S indi-pylibcamera`

I have this error:                                                                                                    

> libcamera-hello: error while loading shared libraries: libboost_program_options.so.1.86.0: cannot open shared object file: No such file or directory

We have the 1.87.0 version of libboost and don't have 1.86 in cache :-(

`ls -l /var/cache/pacman/pkg/boost-* `
-rw-r--r-- 1 root root 15595804 Feb 20 13:04 /var/cache/pacman/pkg/boost-1.87.0-3-aarch64.pkg.tar.xz
-rw-r--r-- 1 root root      566 Feb 20 13:05 /var/cache/pacman/pkg/boost-1.87.0-3-aarch64.pkg.tar.xz.sig
-rw-r--r-- 1 root root  2449628 Feb 20 13:04 /var/cache/pacman/pkg/boost-libs-1.87.0-3-aarch64.pkg.tar.xz
-rw-r--r-- 1 root root      566 Feb 20 13:05 /var/cache/pacman/pkg/boost-libs-1.87.0-3-aarch64.pkg.tar.xz.sig

How I can resolve this issue ?
Thanks

sc74, 2025-04-23:
I don't know if Mattia will update the packages and AstroArch soon. In the meantime, you can look at this other issue to install my libcamera-rpi packages instead of libcamera https://github.com/devDucks/astroarch/issues/142

MattBlack85, 2025-04-25:
April is a pile of scheisse between work and the fact that this week my little one is attending the first communion. Next week I plan to sit down a bit to clean up the long backlog that has piled up during the last months

MattBlack85, 2025-04-30:
@OLEGsandrMB you can try to update now, I shipped a new version of libcamera