devDucks/astroarch #107: cannot update fresh install

albmef, 2024-09-20:
hi
today i started from a fresh install.
i got an error while trying to update the whole distro

sudo pacman -Suy   
:: Synchronizing package databases...
 astromatto is up to date
 core is up to date
 extra is up to date
 community is up to date
 alarm is up to date
 aur is up to date
:: Starting full system upgrade...
...
Total Installed Size:  5212.06 MiB
Net Upgrade Size:       249.99 MiB

:: Proceed with installation? [Y/n] 
:: Retrieving packages...
 kstars-3.7.2-1-aarch64 is up to date
 astro_dmx-2.9.7-1-aarch64 is up to date
 indi-3rdparty-libs-2.0.9-1-aarch64 is up to date
 libindi-2.0.9-1-aarch64 is up to date
 indi-3rdparty-drivers-2.0.9-1-aarch64 is up to date
(553/553) checking keys in keyring                                                                                         [--------------------------------------------------------------------------] 100%
(553/553) checking package integrity                                                                                       [--------------------------------------------------------------------------] 100%
(553/553) loading package files                                                                                            [--------------------------------------------------------------------------] 100%
(553/553) checking for file conflicts                                                                                      [--------------------------------------------------------------------------] 100%
error: failed to commit transaction (conflicting files)
indi-3rdparty-libs: /usr/lib/libqhyccd.so.20 exists in filesystem
Errors occurred, no packages were upgraded.

seems that there is a problem with a very specific symbolic link:  /usr/lib/libqhyccd.so.20 is not owned by any package

~ » pacman -Qo /usr/lib/libqhyccd.so.24.4.1  
/usr/lib/libqhyccd.so.24.4.1 is owned by indi-3rdparty-libs 2.0.8-1
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
~ » pacman -Qo /usr/lib/libqhyccd.so.24     
/usr/lib/libqhyccd.so.24 is owned by indi-3rdparty-libs 2.0.8-1
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
~ » pacman -Qo /usr/lib/libqhyccd.so.20   
error: No package owns /usr/lib/libqhyccd.so.20
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
~ » pacman -Qo /usr/lib/libqhyccd.so  
/usr/lib/libqhyccd.so is owned by indi-3rdparty-libs 2.0.8-1
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
~ »                                                    

so system upgrade cannot proceed

» pacman -Qu | grep indi       
indi-3rdparty-drivers 2.0.8-1 -> 2.0.9-1
indi-3rdparty-libs 2.0.8-1 -> 2.0.9-1
libappindicator-gtk3 12.10.0.r298-3 -> 12.10.0.r298-4
libindi 2.0.8-2 -> 2.0.9-1

regards,
bob

MattBlack85, 2024-09-20:
@albmef I vaguely remember this issue, can you try update-astroarch instead of pacman?

albmef, 2024-09-21:
hi 
you're right i had to to an update-astroarch instead
thanks for your support
bob

closed 2024-09-21 by albmef