devDucks/astroarch #108: dependency break with astrometry.net

albmef, 2024-09-24:
hi 
astrometry.net does not work anymore
it depends on gsl 2.7 but gsl has bumped to v2.8
please rebuild 

~ » solve-field      
solve-field: error while loading shared libraries: libgsl.so.27: cannot open shared object file: No such file or directory

~ » wcsinfo        
wcsinfo: error while loading shared libraries: libgsl.so.27: cannot open shared object file: No such file or directory

~ » pacman -Qi gsl       

Name            : gsl
Version         : 2.8-1
Description     : The GNU Scientific Library (GSL) is a modern numerical library for C and C++ programmers
Architecture    : aarch64
URL             : https://www.gnu.org/software/gsl/gsl.html
Licenses        : GPL-3.0-only
Groups          : None
Provides        : None
Depends On      : cblas  glibc  sh
Optional Deps   : None
Required By     : astrometry.net  libindi  stellarsolver
Optional For    : None
Conflicts With  : None
Replaces        : None
Installed Size  : 4.74 MiB
Packager        : Arch Linux ARM Build System <builder+seattle@archlinuxarm.org>
Build Date      : Sun Jun 9 01:15:35 2024
Install Date    : Tue Jun 11 08:09:24 2024
Install Reason  : Installed as a dependency for another package
Install Script  : No
Validated By    : Signature

best regards,
bob

albmef, 2024-09-25:
i rebuilt from here https://aur.archlinux.org/packages/astrometry.net
changed to arch=('aarch64')
installed manually with pacman -U
seems to be ok so far
i will try an actual platesolving when the sky is clear

~/pkg/astrometry.net (master*) » ldd pkg/astrometry.net/usr/bin/solve-field    
        linux-vdso.so.1 (0x0000ffffbb493000)
        libcfitsio.so.10 => /usr/lib/libcfitsio.so.10 (0x0000ffffbb070000)
        libgsl.so.28 => /usr/lib/libgsl.so.28 (0x0000ffffbad70000)   <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<OK
        libm.so.6 => /usr/lib/libm.so.6 (0x0000ffffbacc0000)
        libwcs.so.8 => /usr/lib/libwcs.so.8 (0x0000ffffbab30000)
        libc.so.6 => /usr/lib/libc.so.6 (0x0000ffffba970000)
        /lib/ld-linux-aarch64.so.1 => /usr/lib/ld-linux-aarch64.so.1 (0x0000ffffbb456000)
        libcurl.so.4 => /usr/lib/libcurl.so.4 (0x0000ffffba870000)
        libz.so.1 => /usr/lib/libz.so.1 (0x0000ffffba840000)
        libcblas.so.3 => /usr/lib/libcblas.so.3 (0x0000ffffba7f0000)
        libnghttp3.so.9 => /usr/lib/libnghttp3.so.9 (0x0000ffffba7a0000)
        libnghttp2.so.14 => /usr/lib/libnghttp2.so.14 (0x0000ffffba750000)
        libidn2.so.0 => /usr/lib/libidn2.so.0 (0x0000ffffba700000)
        libssh2.so.1 => /usr/lib/libssh2.so.1 (0x0000ffffba6a0000)
        libpsl.so.5 => /usr/lib/libpsl.so.5 (0x0000ffffba660000)
        libssl.so.3 => /usr/lib/libssl.so.3 (0x0000ffffba540000)
        libcrypto.so.3 => /usr/lib/libcrypto.so.3 (0x0000ffffb9fd0000)
        libgssapi_krb5.so.2 => /usr/lib/libgssapi_krb5.so.2 (0x0000ffffb9f50000)
        libzstd.so.1 => /usr/lib/libzstd.so.1 (0x0000ffffb9e60000)
        libbrotlidec.so.1 => /usr/lib/libbrotlidec.so.1 (0x0000ffffb9e40000)
        libblas.so.3 => /usr/lib/libblas.so.3 (0x0000ffffb9db0000)
        libunistring.so.5 => /usr/lib/libunistring.so.5 (0x0000ffffb9be0000)
        libkrb5.so.3 => /usr/lib/libkrb5.so.3 (0x0000ffffb9ad0000)
        libk5crypto.so.3 => /usr/lib/libk5crypto.so.3 (0x0000ffffb9a80000)
        libcom_err.so.2 => /usr/lib/libcom_err.so.2 (0x0000ffffb9a50000)
        libkrb5support.so.0 => /usr/lib/libkrb5support.so.0 (0x0000ffffb9a20000)
        libkeyutils.so.1 => /usr/lib/libkeyutils.so.1 (0x0000ffffb99f0000)
        libresolv.so.2 => /usr/lib/libresolv.so.2 (0x0000ffffb99c0000)
        libbrotlicommon.so.1 => /usr/lib/libbrotlicommon.so.1 (0x0000ffffb9980000)
        libgfortran.so.5 => /usr/lib/libgfortran.so.5 (0x0000ffffb97c0000)
        libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x0000ffffb9780000)

MattBlack85, 2024-09-29:
thanks @albmef , this has been fixed in upstream and the next pacman/update-astroarch will fix it

closed 2024-09-29 by MattBlack85