devDucks/astroarch #109: QHY174GPS camera cooling doesn't work

MatiasLM, 2024-09-29:
Im using QHY174GPS camera + HYCFW3 filter wheel connected to a Raspberry Pi 4B 8GB with AstroArch (with INDI and Kstars). Camera and filter wheel controla and adquisición seems to be working well however i'snt possible to control camera temperature.

Even though the set point is set to 0° the camera temperature remains at ambient temperature (no fixed). I tried to do the same from the configuration menu but I get the same results. Even if I set it to automatic mode with Collong Power at 100% the camera remains at room temperature. See attachments.
[image]
[image]

It is worth noting that the camera is powered externally and the cooling system works. Whitout using INDI, If I use the Raspberryas a USB server hub (virtualhere) and start other software such as Shapcap on the client PC, it is possible to control the temperature correctly.

Full hardware configuration
- QHY174GPS camera + HYCFW3 filter wheel connected to camera
- ZWO EAF
- Sky Whatchet HEQ5(EQMOD)
- QHY5III290M (guiding)
- Raspberry Pi 4B 8GB

Full software configuration
- AstroArch SO (https://github.com/devDucks/astroarch)
- Kstars 3.7.2
- indi 2.0.9

MattBlack85, 2024-09-29:
hi @MatiasLM will post on the discord to see if anyone else has issues with QHY. I vaguely remember some old issues with cooling but I don't recall if they were related to QHY or touptek

FireNWater, 2024-09-29:
I have the QHY-600M camera w/filter wheel - similar to your setup.  Sometimes the INID driver will show both buttons "ON" and "OFF" red at the same time while I'm trying to turn on the cooler.  It seems to resolve after a while, though.  Hope this helps.  Are you able to capture any of the INDI log files to help troubleshoot?

MatiasLM, 2024-09-29:
Hi, some updates.
The issue occurs with and without filter wheel connected.

I enabled logging, start INDI, power on and configure the camera cooler both from Ekos capture pamel and INDI control panel. I attach the log file.
[image]
[log_15-09-51.txt](https://github.com/user-attachments/files/17180717/log_15-09-51.txt)

MatiasLM, 2024-10-09:
Hi, 

Unfortunately I can't solve this problem yet. The only advance is that I published the issue en INDI sourceforge forum and a QHY devolver couldn't reproduce the bug and found some strange version for the SDK in the logs.https://sourceforge.net/p/indi/bugs/99/

Anybody could help?

Best regards,

MattBlack85, 2024-10-09:
hi @MatiasLM I am not sure what is that repository on source forge but as far as I know the development happens on github under https://github.com/indilib/indi-3rdparty/

The SDK version is set directly there https://github.com/indilib/indi-3rdparty/blob/master/libqhy/CMakeLists.txt#L5

MattBlack85, 2025-02-06:
hi @MatiasLM, I am following the issue on the other channel although it seems a bit of back and forth. Yesterday I released the latest version of Indi and I am confident they have shipped a new version of the SDK, could you maybe try to update it and check if this version works?

MatiasLM, 2026-02-07:
Hi everyone, sorry for the late reply. I've updated astroarch to the following versions:
- kstars 3.8.0-2
- libindi 2.1.7
- indi-3rdparty-drivers 2.1.7.1-2
- SDK QHY: 25.10.29.6 svn: 1
- libusb: 1.0.29.11953.

It seems the issue has been resolved, as you can see in the screenshot. Same Hardware.

[image]

Does anyone know what the specific fix was?

MattBlack85, 2026-02-08:
Not 100% sure @MatiasLM but maybe the SDK had a bug they fixed in a later version. That would be my bet

MatiasLM, 2026-02-23:
Hi,

@MattBlack85, could you reopen the issue? (I don't have permission.)

Are there any tests I can run to troubleshoot this?

Best regards,

MattBlack85, 2026-02-23:
@MatiasLM are you on latest Indi?

MattBlack85, 2026-09-18:
closed due inactivity

closed 2026-09-18 by MattBlack85