devDucks/astroarch #115: IMX462 sensor with Raspberry Pi 5

Olli781, 2024-10-06:
Hello everyone,
I tried to run the following camera on a Raspberry Pi 5:
innomaker Raspberry Pi industrial camera module, STARVIS IMX462

Due to the built-in sensor, the camera would be well suited for astrophotography.
Unfortunately, the camera is not recognized despite the following entries in the config.txt:
# Automatically load overlays for detected cameras
camera_auto_detect=0
dtoverlay=imx462,clock frequency=74250000

The standard camera with the IMX219 sensor is recognized without errors and works with the INDI Pylibcamera driver perfect.
Is it possible in principle to operate the IMX62 sensor with Astroarch?

Thank you, Oliver

MattBlack85, 2024-10-07:
Hi @Olli781,

sensor support does not depends directly on the distribution but rather if libcamera supports it or not. Do you happen to know if that sensor should be supported out of the box?

Olli781, 2024-10-08:
Hi Mattia,

thank you very much for your reply. According to the manufacturer, libcamera supports the sensor. You just have to adapt the config.txt. Then the sensor should be recognized as IMX290. When reading the log file, the following line is also output:

rp1-cfe 1f00128000.csi: found subdevice /axi/pcie@120000/rp1/i2c@80000/imx290@1a

For better understanding, I have attached the config.txt and the log file.

[config.txt](https://github.com/user-attachments/files/17294281/config.txt)

[Logfile.txt](https://github.com/user-attachments/files/17294284/Logfile.txt)

MattBlack85, 2024-10-08:
hi @Olli781 I had a look at both libcamera and the rpi version of libcamera and I can't find any trace of the imx462, while I can see specific code for the imx290

https://github.com/search?q=repo%3Araspberrypi%2Flibcamera+imx290&type=code
https://github.com/search?q=repo%3Alibcamera-org%2Flibcamera+imx290&type=code

I found this one => https://forums.raspberrypi.com/viewtopic.php?p=1946571#p1946571

can you try to set the overlay directly to imx290?

Olli781, 2024-10-09:
Hi Mattia,

unfortunately, switching overlay directly to imx290 didn't work either. At the same time, I started a request to Innomaker Support. Here is the feedback:

_The  RPI linux kernel using  the same driver for cmos sensor imx290,imx327 and  imx462  so 
the infomation show it  is imx290 sensor  is correct .
From the log file I can see the sensor was successfully registed as /dev/video0 and /dev/video1.
Please try to using :
$libcamera-hello -t 0
have a test and send me the test log._

The command “libcamera-hello -t 0” returned the following message

~ » libcamera-hello -t 0                                                                                                                                                                  astronaut@astroarch
[0:01:13.805134842] [1444] ERROR IPAModule ipa_module.cpp:172 Symbol ipaModuleInfo not found
[0:01:13.805162083] [1444] ERROR IPAModule ipa_module.cpp:292 v4l2-compat.so: IPA module has no valid info
[0:01:13.805206156] [1444]  INFO Camera camera_manager.cpp:313 libcamera v0.3.0
[0:01:13.827660086] [1445]  INFO RPI pisp.cpp:695 libpisp version v1.0.6 15700c36c6c6-dirty 09-09-2024 (14:10:49)
[0:01:13.860931249] [1445] ERROR IPAProxy ipa_proxy.cpp:149 Configuration file 'imx290.json' not found for IPA module 'rpi/pisp'
[0:01:13.861011268] [1445]  WARN RPiController controller.cpp:89 Failed to open tuning file ''
[0:01:13.861020675] [1445] ERROR IPARPI ipa_base.cpp:147 Failed to load tuning data file 
[0:01:13.861028805] [1445] ERROR RPI pipeline_base.cpp:798 Failed to load a suitable IPA library
[0:01:13.861035916] [1445] ERROR RPI pisp.cpp:922 Failed to register camera imx290 4-001a: -22
libEGL warning: DRI3: Screen seems not DRI3 capable
libEGL warning: DRI2: failed to authenticate
libEGL warning: DRI3: Screen seems not DRI3 capable
MESA: error: ZINK: vkCreateInstance failed (VK_ERROR_INCOMPATIBLE_DRIVER)
libEGL warning: egl: failed to create dri2 screen
Made X/EGL preview window
ERROR: *** no cameras available ***

There seems to be a problem with the IPA module.

Best regards, Oliver

MattBlack85, 2024-10-09:
hey @Olli781 can you tell me what version of libpisp libcamera you have?

MattBlack85, 2024-10-09:
libpisp is there, 1.0.6 I see

MattBlack85, 2024-10-14:
hi @Olli781 the file should be available in the new version of libcamera, I am unsure when I will find the time to relase it, but hopefully it will be this week

Olli781, 2025-02-07:
Hi Mattia,
yes the camera works. We can close this topic. Thank you very much!

closed 2025-02-07 by Olli781