devDucks/astroarch #118: QHYCCD cámera exposure timeout

MatiasLM, 2024-10-12:
Hi,

Im using QHY174GPS as main camera (+ HYCFW3 filter wheel) and QHY5III290M for guidance. Both conected to Raspberry Pi 4B 8GB with AstroArch (with INDI and Kstars). 

During a sesión some captures takes several times more that usual due to Exposure timeout. Some nights this ocurre a couple o times but others happens too often that session must be discarded.

Here you can see an example of a real complete session discarded with 360s exposure time.
[image]

Bellow yo can finde a test lesion (without mount and focuser control) for 300s exposure time were I enabled logging.
[image]
[image]
[image]
```
2024-10-12T06:27:31 [QHY CCD QHY174M-1a59dc4] Received image 52 out of 60.
2024-10-12T06:27:31 [QHY CCD QHY174M-1a59dc4] Download Time: 0.22 s, New Download Time Estimate: 0.24 s.
2024-10-12T06:22:29 [QHY CCD QHY174M-1a59dc4] Exposure timeout. Restarting exposure...
2024-10-12T06:14:29 [QHY CCD QHY174M-1a59dc4] Exposure timeout. Restarting exposure...
2024-10-12T06:11:29 [QHY CCD QHY174M-1a59dc4] Capturing 300.000-second Clear image...
```
[log_01-52-44.txt](https://github.com/user-attachments/files/17350762/log_01-52-44.txt)

Full hardware configuration
- QHY174GPS camera + HYCFW3 filter wheel connected to camera
- ZWO EAF
- Sky Whatchet HEQ5(EQMOD)
- QHY5III290M (guiding)
- Raspberry Pi 4 Model B Rev 1.5 8GB

Full software configuration
- AstroArch SO (https://github.com/devDucks/astroarch)
- Kstars 3.7.3 Stable
- INDI Library: 2.1.0 (Code 2.1.0-tgz. Protocol 1.7.)

It should be noted that this also happened to me with  INDI Library: 2.0.9 and Kstars 3.7.2. On the other hand, to rule out setup problems, I have not had this problem using the camera in Windows with the ASCOM driver.

Best Regards,

MattBlack85, 2025-07-10:
@MatiasLM is this still an issue?

MatiasLM, 2026-02-07:
Hi everyone, sorry for the late reply. I've updated astroarch to the following versions:
- kstars 3.8.0-2
- libindi 2.1.7
- indi-3rdparty-drivers 2.1.7.1-2
- SDK QHY: 25.10.29.6 svn: 1
- libusb: 1.0.29.11953.

It seems the issue has been resolved, as you can see in the screenshot. Same Hardware.

[image]

Does anyone know what the specific fix was?