devDucks/astroarch #120: Local language setting

Olli781, 2024-10-28:
Hello,
when I change the language I get the following message in the terminal:

_perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = "de",
        LC_ALL = (unset),
        LC_CTYPE = "en_US.UTF-8",
        LC_NUMERIC = "de_DE.UTF-8",
        LC_COLLATE = (unset),
        LC_TIME = "de_DE.UTF-8",
        LC_MESSAGES = (unset),
        LC_MONETARY = "de_DE.UTF-8",
        LC_ADDRESS = "de_DE.UTF-8",
        LC_IDENTIFICATION = (unset),
        LC_MEASUREMENT = "de_DE.UTF-8",
        LC_PAPER = "de_DE.UTF-8",
        LC_TELEPHONE = "de_DE.UTF-8",
        LC_NAME = "de_DE.UTF-8",
        LANG = "de_DE.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C")._

When starting PHD Guiding the following message appears:

_Cannot set locale to language "German (Germany)"_

The troubleshooting is probably easy. But I can't figure it out right now. Can someone help me?
Thank you, Oliver

MattBlack85, 2024-10-28:
hi @Olli781 , have you already uncommented de_DE local in `/etc/locale.gen` and run the command `locale-gen`?

Olli781, 2024-10-29:
Hi MattBlack85,
I ran the command _sudo locale-gen_ and uncommented de_DE beforehand:
_Generating locales...
  de_DE.UTF-8... done
  de_DE.ISO-8859-1... done
  de_DE.ISO-8859-15@euro... done
Generation complete._

The error with PHD has disappeared. Unfortunately, the error still appears in the terminal:

perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = "de",
        LC_ALL = (unset),
        LC_CTYPE = "en_US.UTF-8",
        LC_NUMERIC = "de_DE.UTF-8",
        LC_COLLATE = (unset),
        LC_TIME = "de_DE.UTF-8",
        LC_MESSAGES = (unset),
        LC_MONETARY = "de_DE.UTF-8",
        LC_ADDRESS = "de_DE.UTF-8",
        LC_IDENTIFICATION = (unset),
        LC_MEASUREMENT = "de_DE.UTF-8",
        LC_PAPER = "de_DE.UTF-8",
        LC_TELEPHONE = "de_DE.UTF-8",
        LC_NAME = "de_DE.UTF-8",
        LANG = "de_DE.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("de_DE.UTF-8").

Olli781, 2024-10-29:
I solved the problem:
en_US.UTF-8 UTF-8 must also remain uncommented.

PR #163 (2025-07-18): Release 2.0
What's new:

- Fixed HDMI issue
- Fixed plasma x11 session issue
- Added visual tool to change settings
- Fixed xorg.conf config

Fix #159 
Fix #161
Fix #155
Fix #133
Fix #129
Fix #120